serialport/node-serialport #435: Cannot install on RPi

solshark, 2015-01-04:
Raspbian 7, armv6l

```
> serialport@1.4.10 install /data/www/aquacontroller-04-01-2015-22:08:53/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: Entering directory '/data/www/aquacontroller-04-01-2015-22:08:53/node_modules/serialport/build'
action_after_build.target.mk:6: *** target pattern contains no '%'.  Stop.
make: Leaving directory '/data/www/aquacontroller-04-01-2015-22:08:53/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:820:12)
gyp ERR! System Linux 3.18.1+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/node-gyp/bin/node-gyp.js" "rebuild" "--name=serialport" "--configuration=Release" "--module_name=serialport" "--version=1.4.10" "--major=1" "--minor=4" "--patch=10" "--runtime=node" "--node_abi=node-v11" "--platform=linux" "--target_platform=linux" "--arch=arm" "--target_arch=arm" "--module_main=./serialport" "--host=https://node-serialport.s3.amazonaws.com/" "--module_path=/data/www/aquacontroller-04-01-2015-22:08:53/node_modules/serialport/build/serialport/v1.4.10/Release/node-v11-linux-arm" "--module=/data/www/aquacontroller-04-01-2015-22:08:53/node_modules/serialport/build/serialport/v1.4.10/Release/node-v11-linux-arm/serialport.node" "--remote_path=./serialport/v1.4.10/Release/" "--package_name=node-v11-linux-arm.tar.gz" "--staged_tarball=build/stage/serialport/v1.4.10/Release/node-v11-linux-arm.tar.gz" "--hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.4.10/Release/" "--hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.4.10/Release/node-v11-linux-arm.tar.gz"
gyp ERR! cwd /data/www/aquacontroller-04-01-2015-22:08:53/node_modules/serialport
gyp ERR! node -v v0.10.35
gyp ERR! node-gyp -v v1.0.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/node-gyp/bin/node-gyp.js rebuild --name=serialport --configuration=Release --module_name=serialport --version=1.4.10 --major=1 --minor=4 --patch=10 --runtime=node --node_abi=node-v11 --platform=linux --target_platform=linux --arch=arm --target_arch=arm --module_main=./serialport --host=https://node-serialport.s3.amazonaws.com/ --module_path=/data/www/aquacontroller-04-01-2015-22:08:53/node_modules/serialport/build/serialport/v1.4.10/Release/node-v11-linux-arm --module=/data/www/aquacontroller-04-01-2015-22:08:53/node_modules/serialport/build/serialport/v1.4.10/Release/node-v11-linux-arm/serialport.node --remote_path=./serialport/v1.4.10/Release/ --package_name=node-v11-linux-arm.tar.gz --staged_tarball=build/stage/serialport/v1.4.10/Release/node-v11-linux-arm.tar.gz --hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.4.10/Release/ --hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.4.10/Release/node-v11-linux-arm.tar.gz' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/node-pre-gyp/lib/util/compile.js:76:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:766:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:833:5)
node-pre-gyp ERR! System Linux 3.18.1+
node-pre-gyp ERR! command "node" "/usr/local/bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /data/www/aquacontroller-04-01-2015-22:08:53/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.35
node-pre-gyp ERR! node-pre-gyp -v v0.6.1
node-pre-gyp ERR! not ok 
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/node-gyp/bin/node-gyp.js rebuild --name=serialport --configuration=Release --module_name=serialport --version=1.4.10 --major=1 --minor=4 --patch=10 --runtime=node --node_abi=node-v11 --platform=linux --target_platform=linux --arch=arm --target_arch=arm --module_main=./serialport --host=https://node-serialport.s3.amazonaws.com/ --module_path=/data/www/aquacontroller-04-01-2015-22:08:53/node_modules/serialport/build/serialport/v1.4.10/Release/node-v11-linux-arm --module=/data/www/aquacontroller-04-01-2015-22:08:53/node_modules/serialport/build/serialport/v1.4.10/Release/node-v11-linux-arm/serialport.node --remote_path=./serialport/v1.4.10/Release/ --package_name=node-v11-linux-arm.tar.gz --staged_tarball=build/stage/serialport/v1.4.10/Release/node-v11-linux-arm.tar.gz --hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.4.10/Release/ --hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.4.10/Release/node-v11-linux-arm.tar.gz' (1)

npm ERR! serialport@1.4.10 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@1.4.10 install script.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 3.18.1+
npm ERR! command "/usr/local/bin/node" "/usr/local/bin/npm" "i" "serialport"
npm ERR! cwd /data/www/aquacontroller-04-01-2015-22:08:53
npm ERR! node -v v0.10.35
npm ERR! npm -v 1.4.28
npm ERR! code ELIFECYCLE
npm ERR! not ok code 0
```

rwaldron, 2015-01-04:
cc @bryan-m-hughes @beriberikix

voodootikigod, 2015-01-04:
Will tackle this tonight (if I can find my raspberry pi -- in the middle of
reconstruction for the house.)

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

bryan-m-hughes, 2015-01-04:
@solshark Can you tell us a little more about the version of Raspbian you're using?

That kernel version seems a little unusual to me. The latest build of Raspbian (2014-12-24) still uses 3.12, not 3.18. Are you using a custom kernal or a custom built version of Raspbian?

Also, what version of node/npm are you using, and how did you install it?

solshark, 2015-01-05:
Sure.
I've used branch "next" for kernel:
```bash
sudo BRANCH=next rpi-update
```

Reason - problems with rfcomm binding. [Details](https://github.com/raspberrypi/linux/issues/683)

Versions:
```
pi@raspberrypi ~ $ node --version
v0.10.35
pi@raspberrypi ~ $ npm --version
1.4.28
```

Installed from [here](http://node-arm.herokuapp.com)

solshark, 2015-01-05:
It might be connected to **node-pre-gyp**
Faced [same issue](https://github.com/mapbox/node-pre-gyp/issues/132) trying to install sqlite3.

bryan-m-hughes, 2015-01-05:
Sounds like gyp is generating a makefile that the next version of make doesn't like. I don't really know what to tell you to fix it unfortunately. Maybe try building it on a different Pi and copy it over? I just tested on my own Pi and node-serialport builds just fine on Raspian stable.

solshark, 2015-01-05:
Strange. I've just downgraded to stable kernel (**System Linux 3.12.35+**) and getting same issue. Maybe I need to do clean installation.

solshark, 2015-01-10:
Updated system using apt-get upgrade and now it works. Unfortunately, have no idea what was the issue - list of updated packages unavailable now.

closed 2015-01-10 by solshark

TomFreudenberg, 2016-06-27:
Reference to: https://github.com/mapbox/node-pre-gyp/issues/222