serialport/node-serialport #439: Compiling for iojs fails

rwaldron, 2015-01-14:
```
$ npm install serialport
|
> serialport@1.4.10 install /Users/rwaldron/wtf/node_modules/serialport
> node-pre-gyp install --fallback-to-build

child_process: customFds option is deprecated, use stdio instead.
child_process: customFds option is deprecated, use stdio instead.
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:481:19: error: no type named 'ExternalAsciiStringResource' in 'v8::String'; did you mean
      'ExternalStringResource'?
      v8::String::ExternalAsciiStringResource *resource) {
      ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
                  ExternalStringResource
/Users/rwaldron/.node-gyp/1.0.1/deps/v8/include/v8.h:2016:19: note: 'ExternalStringResource' declared here
  class V8_EXPORT ExternalStringResource
                  ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:480:36: error: redefinition of 'NanNew'
  NAN_INLINE v8::Local<v8::String> NanNew(
                                   ^
../node_modules/nan/nan.h:475:36: note: previous definition is here
  NAN_INLINE v8::Local<v8::String> NanNew(
                                   ^
../node_modules/nan/nan.h:2103:12: error: no member named 'IsExternalAscii' in 'v8::String'; did you mean 'IsExternal'?
  if (str->IsExternalAscii()) {
           ^~~~~~~~~~~~~~~
           IsExternal
/Users/rwaldron/.node-gyp/1.0.1/deps/v8/include/v8.h:1980:8: note: 'IsExternal' declared here
  bool IsExternal() const;
       ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:2104:23: error: no type named 'ExternalAsciiStringResource' in 'v8::String'; did you mean
      'ExternalStringResource'?
    const v8::String::ExternalAsciiStringResource* ext;
          ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
                      ExternalStringResource
/Users/rwaldron/.node-gyp/1.0.1/deps/v8/include/v8.h:2016:19: note: 'ExternalStringResource' declared here
  class V8_EXPORT ExternalStringResource
                  ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:2105:16: error: no member named 'GetExternalAsciiStringResource' in 'v8::String'; did you mean
      'GetExternalStringResource'?
    ext = str->GetExternalAsciiStringResource();
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
               GetExternalStringResource
/Users/rwaldron/.node-gyp/1.0.1/deps/v8/include/v8.h:6921:41: note: 'GetExternalStringResource' declared here
String::ExternalStringResource* String::GetExternalStringResource() const {
                                        ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:2106:11: error: assigning to 'const char *' from incompatible type 'const uint16_t *'
      (aka 'const unsigned short *')
    *data = ext->data();
          ^ ~~~~~~~~~~~
6 errors generated.
make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1038:12)
gyp ERR! System Darwin 14.0.0
gyp ERR! command "/usr/local/bin/iojs" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild" "--name=serialport" "--configuration=Release" "--module_name=serialport" "--version=1.4.10" "--major=1" "--minor=4" "--patch=10" "--runtime=node" "--node_abi=node-v42" "--platform=darwin" "--target_platform=darwin" "--arch=x64" "--target_arch=x64" "--module_main=./serialport" "--host=https://node-serialport.s3.amazonaws.com/" "--module_path=/Users/rwaldron/wtf/node_modules/serialport/build/serialport/v1.4.10/Release/node-v42-darwin-x64" "--remote_path=./serialport/v1.4.10/Release/" "--package_name=node-v42-darwin-x64.tar.gz" "--staged_tarball=build/stage/serialport/v1.4.10/Release/node-v42-darwin-x64.tar.gz" "--hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.4.10/Release/" "--hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.4.10/Release/node-v42-darwin-x64.tar.gz"
gyp ERR! cwd /Users/rwaldron/wtf/node_modules/serialport
gyp ERR! node -v v1.0.1
gyp ERR! node-gyp -v v1.0.2
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/iojs rebuild --name=serialport --configuration=Release --module_name=serialport --version=1.4.10 --major=1 --minor=4 --patch=10 --runtime=node --node_abi=node-v42 --platform=darwin --target_platform=darwin --arch=x64 --target_arch=x64 --module_main=./serialport --host=https://node-serialport.s3.amazonaws.com/ --module_path=/Users/rwaldron/wtf/node_modules/serialport/build/serialport/v1.4.10/Release/node-v42-darwin-x64 --remote_path=./serialport/v1.4.10/Release/ --package_name=node-v42-darwin-x64.tar.gz --staged_tarball=build/stage/serialport/v1.4.10/Release/node-v42-darwin-x64.tar.gz --hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.4.10/Release/ --hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.4.10/Release/node-v42-darwin-x64.tar.gz' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/Users/rwaldron/wtf/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:76:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:979:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1051:5)
node-pre-gyp ERR! System Darwin 14.0.0
node-pre-gyp ERR! command "node" "/Users/rwaldron/wtf/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /Users/rwaldron/wtf/node_modules/serialport
node-pre-gyp ERR! node -v v1.0.1
node-pre-gyp ERR! node-pre-gyp -v v0.5.19
node-pre-gyp ERR! not ok
Failed to execute '/usr/local/bin/iojs rebuild --name=serialport --configuration=Release --module_name=serialport --version=1.4.10 --major=1 --minor=4 --patch=10 --runtime=node --node_abi=node-v42 --platform=darwin --target_platform=darwin --arch=x64 --target_arch=x64 --module_main=./serialport --host=https://node-serialport.s3.amazonaws.com/ --module_path=/Users/rwaldron/wtf/node_modules/serialport/build/serialport/v1.4.10/Release/node-v42-darwin-x64 --remote_path=./serialport/v1.4.10/Release/ --package_name=node-v42-darwin-x64.tar.gz --staged_tarball=build/stage/serialport/v1.4.10/Release/node-v42-darwin-x64.tar.gz --hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.4.10/Release/ --hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.4.10/Release/node-v42-darwin-x64.tar.gz' (1)
npm ERR! Darwin 14.0.0
npm ERR! argv "node" "/usr/local/bin/npm" "install" "serialport"
npm ERR! node v1.0.1
npm ERR! npm  v2.1.18
npm ERR! code ELIFECYCLE
```

And here's the npm-debug.log: https://gist.github.com/317456f42829a8b6091f

timoxley, 2015-01-15:
https://github.com/iojs/io.js/issues/456

springmeyer, 2015-01-15:
Upgrading nan to 1.5 should fix this one.

rwaldron, 2015-01-15:
[You mean like this?](https://github.com/voodootikigod/node-serialport/pull/441) :P

mikeal, 2015-01-28:
the fix was merged, this should be closed.

voodootikigod, 2015-01-28:
It has been left open until the merge has been verified in target
platforms. Verification is still in process for both io and node support.
the 'merge' is into a yet to be released version.

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

mikeal, 2015-01-28:
ah, that makes sense :)

voodootikigod, 2015-02-11:
1.5.0 published.

closed 2015-02-11 by voodootikigod