serialport/node-serialport #442: serialport.list returns no data

24X7, 2015-01-16:
I have a MacBook Pro running windows bootcamp (does not seem to be the issue though) and it is not able to list any of the ports nor connect.  Am I missing something?

reconbot, 2015-01-16:
Can you give us some sample code? What version of node are you running? And lastly, What does the serialportlist command output? (you might have to find it in your `node-modules/bin` directory)

24X7, 2015-01-16:
Yeah... pretty simple... The array printed is empty. "[]"

require('serialport').list(function (err, results) 
{
    if (err) 
    {
       throw err;
    }
    
    console.log(results);
});

reconbot, 2015-01-16:
The version of node, and the version of windows?

24X7, 2015-01-16:
Node is 0.10.32 and windows is 8.1.  Note I do not have the issue on OSX.

Alex-Werner, 2015-01-19:
I have the same issues on windows 8.1 with Node 0.10.32

voodootikigod, 2015-01-20:
Have you installed the FTDI drivers

24X7, 2015-01-20:
So I have drivers for the devices installed. I did not however install anything from the FTDrivers site.  Is this required if you already have drivers installed?

reconbot, 2015-01-20:
Can anything see the com port?

haydockjp, 2015-01-23:
I am seeing the same thing. I did noticed is that in "serialport.js" Line: 658.  The default (for my platform) is "listUnix".  My "process.platform" equates to "Linux".

Looking at "listUnix" line 510 and 525, '/dev/serial/by-id' is used.  It doesn't look like that is used on (at least my version of) Linux.  It looks like it separates out the device types.

find /dev -name "by-id"
/dev/v4l/by-id
/dev/snd/by-id
/dev/disk/by-id

For reference:

uname -a
Linux nodejs 2.6.32-5-686 #1 SMP Tue May 13 16:33:32 UTC 2014 i686 GNU/Linux

node -v
v0.10.2

reconbot, 2015-01-23:
@24X7 can you run `node -e 'console.log(process.platform)'`

haydockjp, 2015-01-24:
node -e 'console.log(process.platform)'
linux

haydockjp, 2015-01-24:
I also get the same on my raspberry pi

uname -a
Linux raspberrypi 3.12.28+ #709 PREEMPT Mon Sep 8 15:28:00 BST 2014 armv6l GNU/Linux

node -e 'console.log(process.platform)'
linux

reconbot, 2015-01-24:
@haydocjp you probably have a similar but different issue then @24x7

haydockjp, 2015-01-25:
Sorry, I think I should be watching https://github.com/voodootikigod/node-serialport/issues/343

jacobrosenthal, 2015-10-24:
Closed with no response from original poster

closed 2015-10-24 by jacobrosenthal

HexaDragon, 2017-02-17:
I have this same issue on windows 7, with node version 6.9.5 and serialport v4.0.7, 
I cannot list any ports on the machine. I can see the COM4 port in Devicemanager on Win7. 

`var SerialPort = require("serialport");
var util = require("util"), repl = require("repl");

SerialPort.list(function (err, ports) {
  ports.forEach(function(port) {
    console.log(port.comName);
    console.log(port.pnpId);
    console.log(port.manufacturer);
  });
});

var serial_port = new SerialPort("COM4", {baudrate: 57600});

console.log("Writing '*IDN?' to the serial port");
serial_port.write("*IDN?\r");

serial_port.on("data", function (data) {
  console.log("here: "+data);
})
serial_port.on("error", function (msg) {
  console.log("error: "+msg);
})

serial_port.close();`

Also I tried with the following code 

`var Serialport = require("serialport");

var Serialport = new serialport("COM4", {
  baudrate : 57600,
  startbits : 1,
  stopbits : 1,
  parity :  "none",
  parser: Serialport.parsers.readline("\n")
});

Serialport.list(function (err, results) {
  if (err) {
    throw err;
  }

  for (var i = 0; i < results.length; i++) {
    var item = results[i];
    console.log('{comName,-15} {pnpId,-20} {manufacturer}', item);
  }
});`

And finally tried to run the (node serialport-list.js) directly from the bin . It never shows anything.

Can someone please help me ?

reconbot, 2017-02-17:
Can you open a new issue and fill out the template? `serialport-list` is a good indicator if your install and setup are working correctly.