serialport/node-serialport #443: Make native returns consistent

jacobrosenthal, 2015-01-19:
Its currently very hard to finish test coverage, especially around native code which would be handy for new platforms like browser-serialport and for recent talk of a large refactor.

Flush for instance:
In unix we return the result of tcflush
https://github.com/voodootikigod/node-serialport/blob/master/src/serialport_unix.cpp#L695
which is " Upon successful completion, 0 is returned. Otherwise, -1 is returned and errno is set to indicate the error."
In windows nothing is currently returned
https://github.com/voodootikigod/node-serialport/blob/master/src/serialport_win.cpp#L482
but if you did you have access to "If the function succeeds, the return value is nonzero.
If the function fails, the return value is zero. To get extended error information, call GetLastError.

jacobrosenthal, 2015-04-10:
Researching this some more. TLDR want to make unix flush/drain/set deal with their error codes locally like windows does and just fill errorstring. Anyone have a problem with generalizing our hyper specific native error messages?

---------------------------------
UNIX

open loads data->result with file handle and data->errorString with message
"Invalid data bits setting %d", data->dataBits
"Cannot open %s", data->path
"Invalid parity setting %d", data->parity
"Invalid stop bits setting %d", data->stopBits
"Error %s calling ioctl( ..., IOSSIOSPEED, %ld )", strerror(errno), speed )

write loads data->result with bytes written and data->errorString with message
"Error %s calling write(...)", strerror(errno)

close loads data->errorString with message
"Unable to close fd %d, errno: %d", data->fd, errno

list loads data->results with the list, doesnt currently use data->errorString

flush loads data->result with result of tcflush, doesnt currently use data->errorString

drain loads data->result with result of tcdrain, doesnt currently use data->errorString

set loads data->result with result of ioctl, doesnt currently use data->errorString

WINDOWS -ignoring polling

all errors are prefixes to the error code translated to text
%s: File not found, prefix
%s: Invalid handle, prefix
%s: Access denied, prefix
%s: operation aborted, prefix
%s: Unknown error code %d", prefix, errorCode

open loads data->result with file handle and data->errorString with message
"BuildCommDCB"
"SetCommState"
"SetCommTimeouts"
"Opening %s", data->path

write loads data->result with bytes written and data->errorString with message
"Writing to COM port (WriteFile)"
"Writing to COM port (GetOverlappedResult)"

close loads data->errorString with message
"closing connection"

list loads data->results with the list, doesnt currently use data->errorString

flush loads data->errorString with message
"flushing connection"

drain loads data->errorString with message
"draining connection"

set loads data->result with result of SetCommMask, doesnt currently use data->errorString

reconbot, 2016-04-08:
No problem with it

reconbot, 2016-11-05:
The messages of errors are not yet consistent but I'm happy having the specific errors from the underlying platform communicated.

I'm inclined to use error objects to differentiate between different types of errors. With 5x we are now standardized on actually returning errors when an error occurs, no more return codes. All of it is enforced with tests too. =)

I'm going to close this for now as it's made a lot of progress, happy to entertain specific changes.

closed 2016-11-05 by reconbot