serialport/node-serialport #444: When using openImmediately false and a callback, callback is not called

jacobrosenthal, 2015-01-19:
In the docs and tests this case clearly never thought of. I actually need it for a new implementation I'm working on. Just making sure no one has a problem if I fix this. Either that or we need to change the docs.

from docs:
openImmediately (optional)

Attempts to open a connection to the serial port on process.nextTick. The default is true. Set to false to manually call open() at a later time.

callback (optional)

Called when a connection has been opened. The callback should be a function that looks like: function (error) { ... }

jacobrosenthal, 2015-01-19:
ah shit. we're using that callback as successful open. thats why.

jacobrosenthal, 2015-01-21:
If any time travelers find this. 

My issue was needing errors from the callback which is called for errors, but not for success :( We dont use the Factory emitter pattern for browser-serialport so I can't get my errors there. 
@phated pointed out this pattern is known as [releasing zalgo](http://blog.izs.me/post/59142742143/designing-apis-for-asynchrony) and gave me a solution. 

Since in my implementation I'm never going to auto open the port, I can assume the function is synchronous and do this terrible thing:
```
      var serialPort = new SerialPort.SerialPort(path, options, false, function(err){
        done(err);
      });
      done(null)
```
Doom is upon us. Praise Zalgo.

reconbot, 2015-01-21:
Well you are right, we are releasing Zalgo. If we have options errors they fire right away.

reconbot, 2015-01-21:
I'm going to close this in favor of #448

closed 2015-01-21 by reconbot

phated, 2015-01-21:
This should be reopened to discuss a better way to solve this problem, especially if we are going to introduce breaking changes either way.

commit 819c4e10 (2015-01-21): Ensure we call the open callback asynchronously

 - previously called it synchronously on options errors
 - perviously only called asynchronously on open errors
 - update docs to provide more info about the callback

Closes #444
commit 62666b1b (2015-01-21): Ensure we call the open callback asynchronously

 - previously called it synchronously on options errors
 - previously only called asynchronously on open errors
 - update docs to provide more info about the callback

Closes #444
commit 554ecac5 (2015-02-11): closes #444
commit e89767a4 (2015-02-11): Merge pull request #445 from jacobrosenthal/opencallback

closes #444