serialport/node-serialport #446: SerialPort doesn't write the bytes that i want

Arturete, 2015-01-20:
I have done a project with xbees modules for summer. So i have a bit of experience working with serialport but i'm new in this world and i can be doing noob errors.

I use the xbee-api library to build the specific frames that i need and them i send them to the xbee module with the serialport package. The module receive these bytes but it doesn't work correctly. I have check the frames sending them to the xbee by a terminal aplication and everything work perfectly. So i think that the problem only can be because serialport doesn't write the bytes that im requesting to it. I do a console log inside serialport.write() to know the frame that i'm requesting to send and it is ok but the writer is:

 factory.SerialPortBinding.write(this.fd, buffer, function (err, results) {

wich i can't see what is doing. I think the problem is inside that function.

This is my code:
//**************************************************
xbee = require("../node_modules/xbee-api/lib/xbee-api.js")

var SerialPort = require('../node_modules/firmata/node_modules/serialport').SerialPort;

var xbeeAPI = new xbee.XBeeAPI({
    api_mode: 1
});

var serialport = new SerialPort("/dev/ttyUSB0", {
    baudrate: 54000
    //parser: xbeeAPI.rawParser()
});

var newFrame = {    //añadido por arturo 17-12-2014   16:54
    type: 0x00, // xbee_api.constants.FRAME_TYPE.TX_REQUEST_64
    id: 0x52, // optional, nextFrameId() is called per default
    destination64: "0013A200406FB3A1",
    options: 0x00, // optional, 0x00 is default
    data: 0xf9
};

    serialport.open( function(err) {
        if (err) console.log("error opening", err);

        serialport.write(xbeeAPI.buildFrame(newFrame), function (err) {
            if (err) console.log("error writing", err);
        });

        serialport.on("data", function (data) {
            console.log("datos recobidos==>", data);
        });
    });
//**************************************************

I'll thank any kind of help.

fivdi, 2015-02-21:
The specified baudrate of 54000 looks very unconventional. Should it realy work?

reconbot, 2016-03-27:
agreed, but let us know if you figured this out and how you did it!

closed 2016-03-27 by reconbot