serialport/node-serialport #455: Reading Strings does not succeed

Gentlehag, 2015-01-24:
Hi

i've the following code on arduino and node.js (raspberrypi)

But i don't understand the output.

I expected some kind of "Hello world" represenation but can't find it.

what am i doing wrong?

Futermore i plan to transfer some data wich is formated in json style. what would be bestpractice to send and recevie?

many thanks in advance

```arduino
void setup(void)
{
  // start serial port
  Serial.begin(9600);
}

void loop(void)
{
  String foo ="Hello World";
  Serial.print(foo);  
  Serial.write(1);
  Serial.write(0);
  delay(5000);
}
```

On the raspberry pi i've running the folling node.js app

```javascript
var http = require('http');

var SerialPort = require("serialport").SerialPort;
var serialport = new SerialPort("/dev/ttyACM0", { baudrate: 9600 });
serialport.on('open', function () {
    console.log('Serial Port Opend');
    serialport.on('data', function (data) {
        console.log(data[0]);

        console.log(data);
    });
});

http.createServer(function (req, res) {
    res.writeHead(200, { 'Content-Type': 'text/plain' });
    res.end('Hello World\n');
}).listen(1337, '127.0.0.1');
console.log('Server running at http://127.0.0.1:1337/');
    
```

Console Output
```
Server running at http://127.0.0.1:1337/
Serial Port Opend
1
<Buffer 01 01 01>
1
<Buffer 01 01 01 00>
0
<Buffer 00 00 34 34>
1
<Buffer 01>
1
<Buffer 01 01 00 00>
0
<Buffer 00 34 34>
1
<Buffer 01 01 01>
0
<Buffer 00 00 00 34>
52
<Buffer 34>
1
<Buffer 01>
1
<Buffer 01 01 00 00>
0
<Buffer 00 34 34>
1
<Buffer 01 01 01>
0
<Buffer 00 00 00 34>
52
<Buffer 34>
1
<Buffer 01 01 01 00>
0
<Buffer 00 00 34 34>
1
<Buffer 01 01>
1
<Buffer 01 00 00 00>
52
<Buffer 34 34>
1
<Buffer 01 01 01 00>
0
<Buffer 00 00 34 34>
1
<Buffer 01 01>
1
<Buffer 01 00 00 00>
52
<Buffer 34 34>
1
<Buffer 01 01 01>
0
<Buffer 00 00 00 34>
```

fivdi, 2015-02-21:
To help localize the problem, try communicating with the arduino using a utility like `screen` on the Pi to see if that works.

On Raspbian, `screen` needs to be installed manually if it's not already there:
```
sudo apt-get install screen
```
and then:
```
screen /dev/ttyAMA0 9600
```

reconbot, 2016-03-27:
Let us know if this is still an issue `data` is a buffer, so `data.toString()` might help debug this kind of thing.

closed 2016-03-27 by reconbot