serialport/node-serialport #456: Getting "Write again..." printed to console.  Not sure why

jmbldwn, 2015-02-03:
The only instance of this message in the repo is commented out.

Something I changed caused this to happen; Trying to figure out what I did.

Any ideas?

bhaveshgohel, 2015-02-04:
Try searching "console.log" in all files of project folder. [Sublime: Ctrl+shift+F]

jmbldwn, 2015-02-04:
The standard npm install appears to lag what on github, but the real cause was my call to write wasn't a proper buffer or something that would turn into a buffer.  If you give write a number it doesn't appear to send that character;  Maybe it allocates a buffer that size instead?

Anyway, found my fix, but I would like to know when the git repo gets pushed to the standard npm install.

reconbot, 2016-04-03:
It does allocate a buffer that size. We only accept strings and Buffers in `write`.

closed 2016-04-03 by reconbot