serialport/node-serialport #457: Program halts when port not present.

humblewizard, 2015-02-07:
I have this code working when port 2 is connected but when there is no device connected the program halts saying: 

         Throw er;   // Unhandled "error" event
         Error opening \\.\COM2: File not found". 

I have an error handler that is not being called. This code is part of a web server that I would like to have start up and report that the device is not connected if that is in fact the case.

	sp = new com.SerialPort("COM2", {
		baudrate: 4800,
		parser: com.parsers.readline('\r\n')
	});
	sp.on('error', function (data) {
		console.log("Error Handler for Port " + owner.port + ": " + data);
	});
	sp.on('data', function (text) {................}

humblewizard, 2015-02-07:
Adding this to my code fixes my problem:

      process.on('uncaughtException', function(err) {
	      console.log('Caught exception: ' + err);
      });

But my question is now, what does this do?

     sp.on('error', function (data) {
           console.log("Error Handler for Port " + owner.port + ": " + data);
     });

humblewizard, 2015-02-11:
From reading elsewhere, when process.on('uncaughtException'){} is called the stack may be corrupted and the process needs to be restarted so it is no solution to having sp.on('error') working.

humblewizard, 2015-02-13:
Windows 7 box. Same issue as #458

teledemic, 2015-12-14:
I also see an uncaught error on failed port opens in windows 7. The port is real, but not available. The error handler is never called and the error can't be trapped in a try block.

reconbot, 2016-04-03:
I can't confirm this, but I made sure we have tests covering this across all platforms. `on('error')` seems to be working.

closed 2016-04-03 by reconbot