serialport/node-serialport #458: Close and Disconnect event doesn't work properly

sminodonte, 2015-02-09:
When I gets unplugged a device, I cannot find a way to detect this. There is no automatic event...nothing!  I know that it's an old bug but I've tryed with Windows 8.1 and Windows 7 and I've always had the same response namely no event

I have node  v0.12.0
Anybody still have this problem?
thanks

humblewizard, 2015-02-11:
The unplug shows up with:

      process.on('uncaughtException', function(err) {
	      console.log(err);
      });

It will say: "Type error: undefined is not a function".

How to use this to try to reconnect I'm not sure.

humblewizard, 2015-02-11:
I've never been able to get

        serialport.on('error', function(err){
                console.log(err);
                // try to restart port
        }

to fire no matter what happens.

If this would fire when a port is disconnected then it would help to know what to do next.

sminodonte, 2015-02-11:
Thanks for your reply.
My problem is that I can not intercept  any event when I unplug the Arduino usb port. 

Nothing, also with 
      process.on('uncaughtException', function(err) {
           console.log(err);
      });

I have to develop an application where the user don't know nothing about the serial port and they have to connect and disconnect the device at random time and the App have to stay ON and always ready to listen a new device....but if I can not intercepet event I can'd do that application, do u know what I mean? (sorry about my english)

humblewizard, 2015-02-11:
I need to do the same thing and I don't know how to do it either. 
process.on('uncaughtException', function(err) {} is a death warning for the process. I think this is a flaw in serialport that needs to be addressed.

crapthings, 2015-02-13:
i've got same issue with nw.js then i've found usb-detection, but it failed complie on windows.

voodootikigod, 2015-02-13:
Are these all windows only?

sminodonte, 2015-02-13:
The problem is on Windows 7 and Windows 8.1 for me. I've never try on linux

humblewizard, 2015-02-13:
My issue is with Windows 7

humblewizard, 2015-02-15:
Having a similar issue with Ubuntu 14.04

A serial port (Prolific) connected over a USB port is taking data fine. Then when the USB cable is disconnected there is no error generated, data stops, and yet the other parts of the process keep working just fine. In this case the 

      process.on('uncaughtException',function(){})

that is thrown in the Windows 7 case is not thrown in Linux. So in the Linux case we could look for no data received in x seconds to indicate that the port is disconnected and raise a stink about it with the user. It would be nice if the registered p.on('error') function would be called as soon as the USB cable is pulled. Just plugging the cable back in does not restart the connection.

Port Listing:

        com.list(.......);

is able to see the port come and go so I guess we could just keep running the logging every second or two and when the port gets plugged back in then call for the reconnect.

crapthings, 2015-02-15:
+1 for reconnect.

arahlf, 2015-02-21:
+1 for a way to do this

MadLittleMods, 2015-05-31:
:+1: 

`.on('close', cb)` never fires.

 - Windows 8.1
- Node v0.12.1

sword36, 2015-08-19:
I track to data event from the serial port and update variable `lastDataTime`. I use setInterval to call my function of checking every second. Function of checking looks something like this: 
```javascript
if (Date.now() - lastDataTime > 1000) {
    throw "Disconnect";
}
```

jaymes-bearden, 2015-10-04:
I'm having the same issue too. On Windows, supplying options.disconnectedCallback to new SerialPort(...) is not called when the connection is interrupted.

Also, on('close', ...) and on('disconnect', ...) are not called either.

mizraith, 2015-11-06:
Can confirm as well.   Wanted to around with Node.js and some basic serial port communication.  I'm using the basic readdata example with a callback (setTimeout).   If I set the timeout to 1000, I get about 2-3 responses before the port silently closes.   If I set the timeout to 300, I might get more.   Something is timing out, but there is no reporting, no errors, nothing that i can find to log out.

Also -- I created a serialPort.on('close'...)  to reopen the port, but this never gets fired.

Here's the code:
```javascript
var com = require("serialport");

var serialPort = new com.SerialPort("COM23", {
    baudrate: 9600,
    parser: com.parsers.readline('>')
});

serialPort.on('open',function() {
    console.log('Port open');
    setTimeout(serialPort.write("C\r\n"), 300);
});

serialPort.on('data', function(data) {
    console.log("Temp C: " , data);
    setTimeout(serialPort.write("C\r\n"), 300);

});

// This 'close' never fires.  just dies silently.
serialPort.on('close', function() {
    console.log("it got closed...going to reopen");
    serialPort = new com.SerialPort("COM23", {
        baudrate: 9600,
        parser: com.parsers.readline('>')
    });
})
```

Here's the output:
```
Port open
Temp C:  21

Temp C:  21
```

I've tried different ways to set the timeouts, from external to the callback as well, but I get the same problem -- the port just closes.

reconbot, 2016-03-28:
Disconnected happens in a few different places. We also don't emit the event if we provide a handler, nor do consistently call the disconnect logic. On top of that we remove all events if we're disconnected.

Adding this to the backlog to clearly define in #702

reconbot, 2016-04-06:
Disconnect logic is still in a poor state but we've fixed a few crashes and places where errors were not being handled correctly. When it comes to disconnects windows will work a lot easier now in our latest beta.

https://github.com/voodootikigod/node-serialport/issues/733 Please try it out `serialport@2.0.7-beta3` and report back!

reconbot, 2016-05-23:
We're tracking the disconnect issues here #702 

I think we've solved your other issues so I'm going to close this for now. Feel free to reopen.

closed 2016-05-23 by reconbot

keyiis, 2017-03-23:
@humblewizard Having a similar issue with electron,are you fixed?