serialport/node-serialport #459: NodeJS 12 (Cross) Compile Error

mboldisc, 2015-02-09:
I'm having trouble compiling node-serialport for the latest NodeJS 12 release.  I'm performing a cross-compile.  My guess is that the NodeJS APIs have changed.

Here are some logs:
```
| make: Entering directory `/home/workstation/workspace/aep/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/node-red/0.9.1-r18/git/node_modules/serialport/build'
|   CXX(target) Release/obj.target/serialport/src/serialport.o
| In file included from ../src/serialport.h:5:0,
|                  from ../src/serialport.cpp:3:
| ../node_modules/nan/nan.h: In function 'v8::Handle<v8::Value> NanError(const char*)':
| ../node_modules/nan/nan.h:340:38: error: 'New' is not a member of 'v8::String'
|  # define _NAN_ERROR(fun, errmsg) fun(v8::String::New(errmsg))
|                                       ^
| ../node_modules/nan/nan.h:360:13: note: in expansion of macro '_NAN_ERROR'
|      return  _NAN_ERROR(v8::Exception::Error, errmsg);
|              ^
| ../node_modules/nan/nan.h: In function 'void NanThrowError(const char*)':
| ../node_modules/nan/nan.h:345:7: error: 'ThrowException' is not a member of 'v8'
|        v8::ThrowException(_NAN_ERROR(fun, errmsg));                             \
|        ^
| ../node_modules/nan/nan.h:364:5: note: in expansion of macro '_NAN_THROW_ERROR'
|      _NAN_THROW_ERROR(v8::Exception::Error, errmsg);
|      ^
| ../node_modules/nan/nan.h:340:38: error: 'New' is not a member of 'v8::String'
|  # define _NAN_ERROR(fun, errmsg) fun(v8::String::New(errmsg))
|                                       ^
| ../node_modules/nan/nan.h:345:26: note: in expansion of macro '_NAN_ERROR'
|        v8::ThrowException(_NAN_ERROR(fun, errmsg));                             \
|                           ^
| ../node_modules/nan/nan.h:364:5: note: in expansion of macro '_NAN_THROW_ERROR'
|      _NAN_THROW_ERROR(v8::Exception::Error, errmsg);
|      ^
| ../node_modules/nan/nan.h: In function 'void NanThrowError(v8::Handle<v8::Value>)':
| ../node_modules/nan/nan.h:369:5: error: 'ThrowException' is not a member of 'v8'
|      v8::ThrowException(error);
|      ^
| ../node_modules/nan/nan.h: In function 'v8::Handle<v8::Value> NanError(const char*, int)':
| ../node_modules/nan/nan.h:376:53: error: 'New' is not a member of 'v8::String'
|      v8::Local<v8::Value> err = v8::Exception::Error(v8::String::New(msg));
|                                                      ^
| ../node_modules/nan/nan.h:378:14: error: 'New' is not a member of 'v8::String'
|      obj->Set(v8::String::New("code"), v8::Int32::New(errorNumber));
|               ^
| ../node_modules/nan/nan.h:378:65: error: no matching function for call to 'v8::Int32::New(const int&)'
|      obj->Set(v8::String::New("code"), v8::Int32::New(errorNumber));
|                                                                  ^
| ../node_modules/nan/nan.h:378:65: note: candidate is:
| In file included from /home/workstation/.node-gyp/0.12.0/src/node.h:61:0,
|                  from ../node_modules/nan/nan.h:130,
|                  from ../src/serialport.h:5,
|                  from ../src/serialport.cpp:3:
| /home/workstation/.node-gyp/0.12.0/deps/v8/include/v8.h:2012:25: note: static v8::Local<v8::Integer> v8::Integer::New(v8::Isolate*, int32_t)
|    static Local<Integer> New(Isolate* isolate, int32_t value);
|                          ^
| /home/workstation/.node-gyp/0.12.0/deps/v8/include/v8.h:2012:25: note:   candidate expects 2 arguments, 1 provided
| In file included from ../src/serialport.h:5:0,
|                  from ../src/serialport.cpp:3:
| ../node_modules/nan/nan.h: In function 'v8::Handle<v8::Value> NanTypeError(const char*)':
| ../node_modules/nan/nan.h:340:38: error: 'New' is not a member of 'v8::String'
|  # define _NAN_ERROR(fun, errmsg) fun(v8::String::New(errmsg))
|                                       ^
| ../node_modules/nan/nan.h:390:12: note: in expansion of macro '_NAN_ERROR'
|      return _NAN_ERROR(v8::Exception::TypeError, errmsg);
|             ^
| ../node_modules/nan/nan.h: In function 'void NanThrowTypeError(const char*)':
| ../node_modules/nan/nan.h:345:7: error: 'ThrowException' is not a member of 'v8'
|        v8::ThrowException(_NAN_ERROR(fun, errmsg));                             \
|        ^
| ../node_modules/nan/nan.h:394:5: note: in expansion of macro '_NAN_THROW_ERROR'
|      _NAN_THROW_ERROR(v8::Exception::TypeError, errmsg);
|      ^
[... 500 lines cut ...]
| node-pre-gyp ERR! not ok
| Failed to execute '/home/workstation/workspace/aep/build/tmp/sysroots/x86_64-linux/usr/bin/node /home/workstation/workspace/aep/build/tmp/sysroots/x86_64-linux/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js rebuild --name=serialport --configuration=Release --module_name=serialport --version=1.4.2 --major=1 --minor=4 --patch=2 --runtime=node --node_abi=node-v14 --platform=linux --target_platform=linux --arch=arm --target_arch=arm --module_main=./serialport --host=https://node-serialport.s3.amazonaws.com/ --module_path=/home/workstation/workspace/aep/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/node-red/0.9.1-r18/git/node_modules/serialport/build/serialport/v1.4.2/Release/node-v14-linux-arm --module=/home/workstation/workspace/aep/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/node-red/0.9.1-r18/git/node_modules/serialport/build/serialport/v1.4.2/Release/node-v14-linux-arm/serialport.node --remote_path=./serialport/v1.4.2/Release/ --package_name=node-v14-linux-arm.tar.gz --staged_tarball=build/stage/serialport/v1.4.2/Release/node-v14-linux-arm.tar.gz --hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.4.2/Release/ --hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.4.2/Release/node-v14-linux-arm.tar.gz' (1)
| npm ERR! Linux 3.2.0-4-amd64
| npm ERR! argv "/home/workstation/workspace/aep/build/tmp/sysroots/x86_64-linux/usr/bin/node" "/home/workstation/workspace/aep/build/tmp/sysroots/x86_64-linux/usr/bin/npm" "--arch=arm" "--target_arch=arm" "install" "serialport@1.4.2" "nodemailer@0.7.1"
| npm ERR! node v0.12.0
| npm ERR! npm  v2.5.1
| npm ERR! code ELIFECYCLE
| 
| npm ERR! serialport@1.4.2 install: `node-pre-gyp install --fallback-to-build`
| npm ERR! Exit status 1
| npm ERR!
| npm ERR! Failed at the serialport@1.4.2 install script 'node-pre-gyp install --fallback-to-build'.
| npm ERR! This is most likely a problem with the serialport package,
| npm ERR! not with npm itself.
| npm ERR! Tell the author that this fails on your system:
| npm ERR!     node-pre-gyp install --fallback-to-build
| npm ERR! You can get their info via:
| npm ERR!     npm owner ls serialport
| npm ERR! There is likely additional logging output above.
| 
| npm ERR! Please include the following file with any support request:
| npm ERR!     /home/workstation/workspace/aep/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/node-red/0.9.1-r18/git/npm-debug.log
| ERROR: oe_runnpm failed
| WARNING: exit code 1 from a shell command.
```

imyller, 2015-02-11:
I can also confirm this issue with Node.js 0.12.0 and node-serialport.

imyller, 2015-02-11:
I think issue #454 is related

mboldisc, 2015-02-11:
I think this was a version problem.  I was trying to compile using serialport 1.4.2.  With 1.4.10 it seems to compile.  I wasn't able to test it because of other issues with NodeJS 12 and ARMv5.

voodootikigod, 2015-02-11:
Please try latest 1.5.0 should work fine.

closed 2015-02-11 by voodootikigod