serialport/node-serialport #461: Convert Slow Buffer to String

deliverymanager, 2015-02-19:
I use
serialPort.on('data', function(buffer) {
     console.log(buffer);
     console.log(buffer.toString("utf-8"));
}
I am testing it by making a call to the GSM modem.
but on my cmd console it prints:
<SlowBuffer 32 04 eb 10 fc>
2@?@?

The second line should be the converted string.
What am I doing wrong?
I am not very familiar with converting the buffer to a string.

humblewizard, 2015-02-20:
I think that the type passed in the serialPort.on is an object with a toString method that does not necessarily accept the argument of character type that a buffer object does. Your buffer variable is not a type "buffer" as in:    var buffer = new Buffer(8);

Rather it is a different type of object. Try it without the argument.

deliverymanager, 2015-02-21:
I have tried it without the utf-8 argument, but still no luck.
The result is the same.
I also thought that it might be the encoding that the cmd is using, the reason that I cannot read it, but then I tried printing it to a log file, and the result was the same.
The converted chars were unreadable.
Any other ideas?

fivdi, 2015-02-21:
The `Buffer` in the above example doesn't appear to contain a valid/readable string. If it doesn't contain the string you expect it to contain, then I'd imagine that something went wrong before the `data` event handler was called.

A `Buffer` is used to hold binary data, in the case above, the following bytes: `0x32, 0x04, 0xeb, 0x10, 0xfc`.

The v0.12 documentation for `buf.toSting` is [here](http://nodejs.org/api/buffer.html#buffer_buf_tostring_encoding_start_end)

Here's what my system prints when toString is called on the buffer with the supported encodings:

```
$ node
> var buf = new Buffer([0x32, 0x04, 0xeb, 0x10, 0xfc]);
undefined
> buf
<Buffer 32 04 eb 10 fc>
> buf.toString('ascii');
'2\u0004k\u0010|'
> buf.toString('utf8');
'2\u0004�\u0010�'
> buf.toString('utf16le');
'вძ'
> buf.toString('ucs2');
'вძ'
> buf.toString('base64');
'MgTrEPw='
> buf.toString('binary');
'2\u0004ë\u0010ü'
> buf.toString('hex');
'3204eb10fc'
> 
```

closed 2016-04-03 by reconbot