serialport/node-serialport #465: Read Stream Serial (FLOAT Numbers)

kdsasuke, 2015-02-26:
Hello everyone, 

I'm sending from STM32 a Stream Serial numbers, there are float numbers, I don't know if possible with serialport to read and parse this kind of numbers.

Thanks!

reconbot, 2015-02-26:
It is but you'll need some sort of protocol. You could send them as ASCII with newline delimiters. Or dream up some sort of binary protocol. The first suggestion is a lot easier to debug.

kdsasuke, 2015-02-26:
I'm already send at least 100 float numbers through the Serial Stream from STM32, The numbers was sent in Binary format, so in nodejs I saw things like this:

<Buffer 00 00 00 00 00 00 00 00 00 00 00 00 14 00 00 00 14 00 00 00 14 00 00 00 
00 00 00 00 00 00 00 00> 

And the idea is parse this buffer to Float Numbers. 

Thanks!

reconbot, 2015-02-26:
You need some way to know when a number ends. You can't guarantee that
they'll be sent or received all at once.

closed 2016-04-04 by reconbot