serialport/node-serialport #466: Cannot install on udoochild_process: customFds option is deprecated, use stdio instead.

Willjobs94, 2015-02-26:


fivdi, 2015-03-01:
The message "customFds option is deprecated, use stdio instead." is a deprecation notice and doesn't necessarily mean that there's a problem. What else is output to the terminal during the install process?

Willjobs94, 2015-03-02:
Thanks for your Answer I really appreciate it.
Here's my error!
[image]

I guess is because of the "-pre" in the version, but I don't know how to install another with not "-pre".

fivdi, 2015-03-02:
Does [this](https://github.com/TooTallNate/node-gyp/wiki/Error%3A-%22pre%22-versions-of-node-cannot-be-installed) help?

Willjobs94, 2015-03-03:
Well it does!, now I'm getting another error :), I think I succeeded  the previous error.
I was also able to install it on another udoo with node version v0.12 without problem.
I would like to know the answer of the following error, and also how to replace my node version v0.13.0-pre with the v0.12 instead. Thanks for helping me !!1.

The other error
[image]

closed 2016-04-04 by reconbot