serialport/node-serialport #467: serialport.list not finding some ports.

humblewizard, 2015-02-26:
No doubt I've got something wrong here. There are two USB Prolific port and serialport.list finds one of those. Here are a couple of command line outputs showing 8 serial ports and then there is the output from serialport.list.

dmesg output ******************************************************************
$ dmesg | grep tty
[    0.000000] console [tty0] enabled
[    1.157126] 00:05: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
[    1.177815] 00:06: ttyS1 at I/O 0x2f8 (irq = 3, base_baud = 115200) is a 16550A
[    1.204666] 0000:02:00.0: ttyS4 at I/O 0x1018 (irq = 18, base_baud = 115200) is a ST16650V2
[    1.226622] 0000:02:00.1: ttyS5 at I/O 0x1010 (irq = 19, base_baud = 115200) is a ST16650V2
[    1.248589] 0000:02:00.2: ttyS6 at I/O 0x1008 (irq = 16, base_baud = 115200) is a ST16650V2
[    1.270547] 0000:02:00.3: ttyS7 at I/O 0x1000 (irq = 17, base_baud = 115200) is a ST16650V2
[17004.557423] usb 3-1: pl2303 converter now attached to ttyUSB0
[17009.695468] usb 3-2: pl2303 converter now attached to ttyUSB1

setserial output *******************************************************************
$ sudo setserial -g /dev/tty*
/dev/ttyS0, UART: 16550A, Port: 0x03f8, IRQ: 4
/dev/ttyS1, UART: 16550A, Port: 0x02f8, IRQ: 3
/dev/ttyS4, UART: 16650V2, Port: 0x1018, IRQ: 18
/dev/ttyS5, UART: 16650V2, Port: 0x1010, IRQ: 19
/dev/ttyS6, UART: 16650V2, Port: 0x1008, IRQ: 16
/dev/ttyS7, UART: 16650V2, Port: 0x1000, IRQ: 17
/dev/ttyUSB0, UART: 16654, Port: 0x0000, IRQ: 0
/dev/ttyUSB1, UART: 16654, Port: 0x0000, IRQ: 0

List ports output *********************************************************************

sudo nodejs /home/bob/TBinst/listPorts.js
/dev/ttyUSB1
usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0 Prolific_Technology_Inc.

List ports code ***********************************************************************
      var serialport = require("serialport");
      serialport.list(function (err, ports) {
	  if(err) console.log(err);
          ports.forEach(function(port) {
   	       console.log(port.comName);
               console.log(port.pnpId);
               console.log(port.manufacturer);
          });
      });

rexfm, 2015-04-02:
I'm having the same issue with multiple Arduino clones that use the same USB chip, just showing up as one serial port in sp.list()

I'll let you know if I find a fix, otherwise what I'm planning to do is try to work around the issue with another usb project (node-udev) which I'm already using for more reliable usb event monitoring (plugging and unplugging arduinos) and then open those directly in node-serialport by devName

> $ node scripts/usblist.js | grep "/dev/tty"
    DEVNAME: '/dev/ttyACM0',
    DEVNAME: '/dev/ttyUSB0',
    DEVNAME: '/dev/ttyUSB1',

jesade, 2015-04-17:
On mac, I use this quick workaround...
var execSync = require('exec-sync');
function getPortList(){
	var stdout = execSync("(ls -1S /dev/tty.* ; ls -1S /dev/cu.*)");
	return stdout.split('\n');
}

jacobrosenthal, 2015-10-24:
Looks like the devices are identical, were seeing this issue and working on it https://github.com/voodootikigod/node-serialport/issues/610

closed 2015-10-24 by jacobrosenthal