serialport/node-serialport #468: failed to compile in macos

jemos, 2015-03-01:
Hi,

It was working a few weeks back, but now I get:

...
> node-pre-gyp install --fallback-to-build

child_process: customFds option is deprecated, use stdio instead.
child_process: customFds option is deprecated, use stdio instead.
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
../src/serialport_unix.cpp: In function ‘void EIO_Open(uv_work_t*)’:
../src/serialport_unix.cpp:146: error: ‘O_CLOEXEC’ was not declared in this scope
../src/serialport_unix.cpp: In function ‘void ExtractUsbInformation(stSerialDevice*, IOUSBDeviceInterface**)’:
../src/serialport_unix.cpp:456: warning: format ‘%08x’ expects type ‘unsigned int’, but argument 4 has type ‘UInt32’
make: *** [Release/obj.target/serialport/src/serialport_unix.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1067:12)
gyp ERR! System Darwin 10.8.0

Any idea?

Thank you.
Jean Mousinho

jacobrosenthal, 2015-03-02:
Hi Jean,

Can you get really specific on what you're trying to do? Is this just an npm i? What version node, node-serialport. Is the code on github you can link to so we can try?

Thanks,
Jacob

jemos, 2015-03-02:
Hi Jacob,

Sorry for the lack of information.

I can provide more information when I get home, but I can add the following
for now:

- When installing it will try to compile the code (is this a problem?) as
you probably noticed, but the source is using O_CLOEXEC constant which does
not exist in MacOS headers (Snow Leopard?).

- Trying with P_CLOEXEC makes compiling work, so module installs
successfully afterwards.

- Not sure if its functional since first experiment with a bluetooth serial
device did not work. Need to test with other device.

Thanks.

Regards,
Jean Mousinho

jemos, 2015-03-02:
Regarding the versions I think node is 0.12 and node-serialport is a clone
for the code compilation test and when using npm directly is version 1.5.0.

Regards,
Jean Mousinho

jacobrosenthal, 2015-04-09:
You mention Snow Leopard, is that your version of osx?

jemos, 2015-04-09:
Yes, OSX version 10.6.8

jacobrosenthal, 2015-04-09:
That constant was added in https://github.com/voodootikigod/node-serialport/commit/5671b98815f1db16251ba2543201d7c58592d240 Try to lock to 1.4.6 or below if there are no features you need.

jacobrosenthal, 2015-04-09:
Im not up on my handles so don't actually know if we even need O_CLOEXEC, or what it would have had to do with that polling commit. Perhaps @sweetpi can comment? 

* We could desupport Snow Leopard, I don't know what we want our lowest osx target should be. 
* We can do OSX version checks, which apparently isnt that easy?
* Or just use the old style, dont pass O_CLOEXEC as a open flag, but manually set FD_CLOEXEC with fcntl right after opening.

sweetpi, 2015-04-10:
> Im not up on my handles so don't actually know if we even need O_CLOEXEC, or what it would have had to do with that polling commit. Perhaps @sweetpi can comment?

It's not directly related to polling, more to fixing disconnect crashes. I'm not 100% sure if the flag is needed.

> Or just use the old style, dont pass O_CLOEXEC as a open flag, but manually set FD_CLOEXEC with fcntl right after opening.

I think this should do the job.

closed 2015-09-12 by jacobrosenthal