serialport/node-serialport #469: Problem during installation

shenron, 2015-03-01:
Hi,

I have:
- Raspberry pi B rev1 (ram 256mo)
- ArchLinux
- Node v0.10.28 (last version arm I found)
- npm 1.4.9

During global installation
> npm install -g serialport
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/0.10.28"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/lib/node_modules/serialport/.node-gyp"
gyp http GET http://nodejs.org/dist/v0.10.28/node-v0.10.28.tar.gz
gyp http 200 http://nodejs.org/dist/v0.10.28/node-v0.10.28.tar.gz
gyp http GET http://nodejs.org/dist/v0.10.28/SHASUMS.txt
gyp http 200 http://nodejs.org/dist/v0.10.28/SHASUMS.txt
gyp WARN install got an error, rolling back install
gyp WARN install got an error, rolling back install
gyp ERR! configure error 
gyp ERR! stack Error: node-v0.10.28.tar.gz local sha1 5e3c6e2be25233008125e2a3355a78e654ad27c6 not match remote ef08a75f6359a16e672cae684e0804ca7f4554b7
gyp ERR! stack     at deref (/opt/node/node-v0.10.28-linux-arm-pi/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:296:20)
gyp ERR! stack     at IncomingMessage.<anonymous> (/opt/node/node-v0.10.28-linux-arm-pi/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:336:13)
gyp ERR! stack     at IncomingMessage.EventEmitter.emit (events.js:117:20)
gyp ERR! stack     at _stream_readable.js:919:16
gyp ERR! stack     at process._tickCallback (node.js:419:13)
gyp ERR! System Linux 3.18.8-1-ARCH
gyp ERR! command "/opt/node/node-v0.10.28-linux-arm-pi/bin/node" "/opt/node/node-v0.10.28-linux-arm-pi/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild" "--name=serialport" "--configuration=Release" "--module_name=serialport" "--version=1.5.0" "--major=1" "--minor=5" "--runtime=node" "--node_abi=node-v11" "--platform=linux" "--target_platform=linux" "--arch=arm" "--target_arch=arm" "--module_main=./serialport" "--host=https://node-serialport.s3.amazonaws.com/" "--module_path=/usr/lib/node_modules/serialport/build/serialport/v1.5.0/Release/node-v11-linux-arm" "--remote_path=./serialport/v1.5.0/Release/" "--package_name=node-v11-linux-arm.tar.gz" "--staged_tarball=build/stage/serialport/v1.5.0/Release/node-v11-linux-arm.tar.gz" "--hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.5.0/Release/" "--hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.5.0/Release/node-v11-linux-arm.tar.gz"
gyp ERR! cwd /usr/lib/node_modules/serialport
gyp ERR! node -v v0.10.28
gyp ERR! node-gyp -v v0.13.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/opt/node/node-v0.10.28-linux-arm-pi/bin/node rebuild --name=serialport --configuration=Release --module_name=serialport --version=1.5.0 --major=1 --minor=5 --runtime=node --node_abi=node-v11 --platform=linux --target_platform=linux --arch=arm --target_arch=arm --module_main=./serialport --host=https://node-serialport.s3.amazonaws.com/ --module_path=/usr/lib/node_modules/serialport/build/serialport/v1.5.0/Release/node-v11-linux-arm --remote_path=./serialport/v1.5.0/Release/ --package_name=node-v11-linux-arm.tar.gz --staged_tarball=build/stage/serialport/v1.5.0/Release/node-v11-linux-arm.tar.gz --hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.5.0/Release/ --hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.5.0/Release/node-v11-linux-arm.tar.gz' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:76:29)
node-pre-gyp ERR! stack     at ChildProcess.EventEmitter.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:753:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:820:5)
node-pre-gyp ERR! System Linux 3.18.8-1-ARCH
node-pre-gyp ERR! command "node" "/usr/lib/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/lib/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.28
node-pre-gyp ERR! node-pre-gyp -v v0.5.19
node-pre-gyp ERR! not ok 
Failed to execute '/opt/node/node-v0.10.28-linux-arm-pi/bin/node rebuild --name=serialport --configuration=Release --module_name=serialport --version=1.5.0 --major=1 --minor=5 --runtime=node --node_abi=node-v11 --platform=linux --target_platform=linux --arch=arm --target_arch=arm --module_main=./serialport --host=https://node-serialport.s3.amazonaws.com/ --module_path=/usr/lib/node_modules/serialport/build/serialport/v1.5.0/Release/node-v11-linux-arm --remote_path=./serialport/v1.5.0/Release/ --package_name=node-v11-linux-arm.tar.gz --staged_tarball=build/stage/serialport/v1.5.0/Release/node-v11-linux-arm.tar.gz --hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.5.0/Release/ --hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.5.0/Release/node-v11-linux-arm.tar.gz' (1)
npm ERR! serialport@1.5.0 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@1.5.0 install script.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 3.18.8-1-ARCH
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "install" "-g" "serialport"
npm ERR! cwd /home/shenron/scripts/rapiro
npm ERR! node -v v0.10.28
npm ERR! npm -v 1.4.9
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/shenron/scripts/rapiro/npm-debug.log

I don t know how I can resolve this.

shenron, 2015-03-02:
I use node v0.10.12, with serialport@1.4.0 and now I have these errors:

> serialport@1.4.0 install /opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp http GET https://node-serialport.s3.amazonaws.com/serialport/v1.4.0/Release/node-v11-linux-arm.tar.gz
node-pre-gyp http 404 https://node-serialport.s3.amazonaws.com/serialport/v1.4.0/Release/node-v11-linux-arm.tar.gz
node-pre-gyp http Pre-built binary not available for your system (falling back to source compile with node-gyp) 
gyp http GET http://nodejs.org/dist/v0.10.12/node-v0.10.12.tar.gz
gyp http 200 http://nodejs.org/dist/v0.10.12/node-v0.10.12.tar.gz
gyp WARN install got an error, rolling back install
gyp ERR! configure error 
gyp ERR! stack Error: unexpected eof
gyp ERR! stack     at decorate (/opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/npm/node_modules/fstream/lib/abstract.js:67:36)
gyp ERR! stack     at Extract.Abstract.error (/opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/npm/node_modules/fstream/lib/abstract.js:61:12)
gyp ERR! stack     at Extract._streamEnd (/opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/npm/node_modules/tar/lib/extract.js:75:22)
gyp ERR! stack     at BlockStream.<anonymous> (/opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/npm/node_modules/tar/lib/parse.js:50:8)
gyp ERR! stack     at BlockStream.EventEmitter.emit (events.js:92:17)
gyp ERR! stack     at BlockStream._emitChunk (/opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/npm/node_modules/block-stream/block-stream.js:203:10)
gyp ERR! stack     at BlockStream.resume (/opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/npm/node_modules/block-stream/block-stream.js:58:15)
gyp ERR! stack     at Extract.Reader.resume (/opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/npm/node_modules/fstream/lib/reader.js:252:34)
gyp ERR! stack     at DirWriter.ondrain (stream.js:61:14)
gyp ERR! stack     at DirWriter.EventEmitter.emit (events.js:92:17)
gyp ERR! System Linux 3.18.8-1-ARCH
gyp ERR! command "/opt/node/node-v0.10.12-linux-arm-pi/bin/node" "/opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild" "--name=serialport" "--configuration=Release" "--module_name=serialport" "--version=1.4.0" "--major=1" "--minor=4" "--runtime=node" "--node_abi=node-v11" "--platform=linux" "--target_platform=linux" "--arch=arm" "--target_arch=arm" "--module_main=./serialport" "--host=https://node-serialport.s3.amazonaws.com/" "--module_path=/opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/serialport/build/serialport/v1.4.0/Release/node-v11-linux-arm" "--module=/opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/serialport/build/serialport/v1.4.0/Release/node-v11-linux-arm/serialport.node" "--remote_path=./serialport/v1.4.0/Release/" "--package_name=node-v11-linux-arm.tar.gz" "--staged_tarball=build/stage/serialport/v1.4.0/Release/node-v11-linux-arm.tar.gz" "--hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.4.0/Release/" "--hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.4.0/Release/node-v11-linux-arm.tar.gz"
gyp ERR! cwd /opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/serialport
gyp ERR! node -v v0.10.12
gyp ERR! node-gyp -v v0.10.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/opt/node/node-v0.10.12-linux-arm-pi/bin/node /opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js rebuild --name=serialport --configuration=Release --module_name=serialport --version=1.4.0 --major=1 --minor=4 --runtime=node --node_abi=node-v11 --platform=linux --target_platform=linux --arch=arm --target_arch=arm --module_main=./serialport --host=https://node-serialport.s3.amazonaws.com/ --module_path=/opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/serialport/build/serialport/v1.4.0/Release/node-v11-linux-arm --module=/opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/serialport/build/serialport/v1.4.0/Release/node-v11-linux-arm/serialport.node --remote_path=./serialport/v1.4.0/Release/ --package_name=node-v11-linux-arm.tar.gz --staged_tarball=build/stage/serialport/v1.4.0/Release/node-v11-linux-arm.tar.gz --hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.4.0/Release/ --hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.4.0/Release/node-v11-linux-arm.tar.gz' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:76:29)
node-pre-gyp ERR! stack     at ChildProcess.EventEmitter.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:735:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:802:5)
node-pre-gyp ERR! System Linux 3.18.8-1-ARCH
node-pre-gyp ERR! command "node" "/opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.12
node-pre-gyp ERR! node-pre-gyp -v v0.5.31
node-pre-gyp ERR! not ok 
Failed to execute '/opt/node/node-v0.10.12-linux-arm-pi/bin/node /opt/node/node-v0.10.12-linux-arm-pi/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js rebuild --name=serialport --configuration=Release --module_ [... cut ...] odules/serialport/build/serialport/v1.4.0/Release/node-v11-linux-arm/serialport.node --remote_path=./serialport/v1.4.0/Release/ --package_name=node-v11-linux-arm.tar.gz --staged_tarball=build/stage/serialport/v1.4.0/Release/node-v11-linux-arm.tar.gz --hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.4.0/Release/ --hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.4.0/Release/node-v11-linux-arm.tar.gz' (1)
npm ERR! weird error 1
npm ERR! not ok code 0

Idea ?

shenron, 2015-03-03:
I install a fresh Raspbian and I have the same error with the last version of serialport.

DrCord, 2015-03-15:
I am also having an issue with the latest version of serialport.

voodootikigod, 2015-03-15:
can someone output the results of 

```
npm info debug
```

marcuswinkler, 2015-03-18:
Please see also #489. Installing with a version of node older than v0.10.16 will not work.

closed 2015-09-14 by jacobrosenthal