serialport/node-serialport #472: this.fd = 0 treated as an error

ecksun, 2015-03-02:
I just noticed that when the file descriptor returned from SerialPortBinding.open is 0 it is considered an error.

In most places the fd is checked with truthines, such as for example in the drain method:

```javascript
var fd = this.fd;
if (!fd) {
    // ...
}
```

In afterRead for example a check is made that `self.fd >= 0`, which suggest that 0 is in fact a valid file descriptor.

So my first question then is if 0 is a valid file descriptor?

In my experiments its seems to be, as I get a file descriptor of 0 from SerialPortBinding.open and if I fix all the checks I can communicate with my serial devices.

Secondly I think the checks for `self.fd >= 0` are broken, as if self.fd is null the expression will evaluate to true.

If I have correctly understood the problems I will submit a pull-request fixing the issues

reconbot, 2016-03-27:
It's my understanding that fd0 will always be stdin even if you've closed it. I support this change, but I'm curious has this caused a bug for you? And if so can you reproduce it?

ecksun, 2016-03-27:
Yes, my system does not work without this patch. I could reproduce it rather reliably but I never tried to reduce it to an easy example. It was a long time ago I found and fixed it so I don't remember much about how I found it.

Afaik the fd you get is defined by the kernel. Are you looking for a unit-test to show that this works? A proper system-test would probably require us to mess with the kernel or to mock the open system call

reconbot, 2016-03-28:
I pushed your changes in #703 , if you can still test this situation, I'd love to have you try it. (What operating system and device was it?)

ecksun, 2016-03-28:
I will see if I can get some time. This was several ARM systems running debian stable

reconbot, 2016-03-29:
You'll wan to test `serialport@2.0.7-beta2` on #688

ecksun, 2016-03-29:
Runnig 2.0.6 gives me this:

    nodejs: ../deps/uv/src/unix/core.c:701: uv__io_stop: Assertion `loop->watchers[w->fd] == w' failed.

While 2.0.7-beta2 works fine.

closed 2016-03-29 by ecksun

reconbot, 2016-03-29:
Whoop!

Thanks for fixing this! There will be at least one more beta. I cannot wait for this release.