serialport/node-serialport #475: Error Transport endpoint is not connected calling write

alexcroox, 2015-03-05:
I'm trying to get a Raspberry Pi to connect to an Ardiuno using a linvor bluetooth module.

Using the example from the readme:

```
var SerialPort = require("serialport").SerialPort
var serialPort = new SerialPort("/dev/rfcomm1", {
    baudrate: 57600
});

serialPort.on("open", function() {
    console.log('open');
    serialPort.on('data', function(data) {
        console.log('data received: ' + data);
    });
    serialPort.write("ls\n", function(err, results) {
        console.log('err ' + err);
        console.log('results ' + results);
    });
});
```

The output is:

```
open
err Error: Error Transport endpoint is not connected calling write(...)
results undefined
```

So it looks like it connects OK, but then fails as soon as it tries to write.

However I did notice that the flashing blue light on the module goes solid a few seconds later indicating an established connection. If I alter the above code to include a timeout of 5 seconds before calling .write() it goes through without issue.

Should I be putting in my own timeout before doing any writing to the comm port? I assumed "open" event wouldn't trigger before a connection was established.

alexcroox, 2015-03-09:
Am I misunderstanding the open event and treating it as connected when it's not? The documentation doesn't mention about needing the timeout for it to work, has anyone else attempting to write as soon as the open event is called?

reconbot, 2016-04-04:
It means the port is open, it doesn't guarantee anything about the receiving end, but I think your pi isn't correctly configured. 
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=95197

I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to reopen.

closed 2016-04-04 by reconbot