serialport/node-serialport #477: Error installing on ARM Docker image with io.js 1.4.3

techniq, 2015-03-06:
I'm trying to `npm install serialport` using Docker with an [ARM-based image](https://registry.hub.docker.com/u/strictlybusiness/beaglebone-black-iojs/) and qemu, but the node-gyp compilation is erroring

```
root@73f15e99a1de:/# npm install serialport
-
> serialport@1.6.1 install /node_modules/serialport
> node-pre-gyp install --fallback-to-build

child_process: customFds option is deprecated, use stdio instead.
child_process: customFds option is deprecated, use stdio instead.
make: Entering directory '/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/serialport_poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
collect2: fatal error: vfork: Cannot allocate memory
compilation terminated.
serialport.target.mk:112: recipe for target 'Release/obj.target/serialport.node' failed
make: *** [Release/obj.target/serialport.node] Error 1
make: Leaving directory '/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:169:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1044:12)
gyp ERR! System Linux 3.13.0-40-generic
gyp ERR! command "/usr/bin/iojs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild" "--name=serialport" "--configuration=Release" "--module_name=serialport" "--version=1.6.1" "--major=1" "--minor=6" "--patch=1" "--runtime=node" "--node_abi=node-v43" "--platform=linux" "--target_platform=linux" "--arch=arm" "--target_arch=arm" "--module_main=./serialport" "--host=https://node-serialport.s3.amazonaws.com/" "--module_path=/node_modules/serialport/build/serialport/v1.6.1/Release/node-v43-linux-arm" "--remote_path=./serialport/v1.6.1/Release/" "--package_name=node-v43-linux-arm.tar.gz" "--staged_tarball=build/stage/serialport/v1.6.1/Release/node-v43-linux-arm.tar.gz" "--hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.6.1/Release/" "--hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.6.1/Release/node-v43-linux-arm.tar.gz"
gyp ERR! cwd /node_modules/serialport
gyp ERR! node -v v1.4.3
gyp ERR! node-gyp -v v1.0.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/iojs rebuild --name=serialport --configuration=Release --module_name=serialport --version=1.6.1 --major=1 --minor=6 --patch=1 --runtime=node --node_abi=node-v43 --platform=linux --target_platform=linux --arch=arm --target_arch=arm --module_main=./serialport --host=https://node-serialport.s3.amazonaws.com/ --module_path=/node_modules/serialport/build/serialport/v1.6.1/Release/node-v43-linux-arm --remote_path=./serialport/v1.6.1/Release/ --package_name=node-v43-linux-arm.tar.gz --staged_tarball=build/stage/serialport/v1.6.1/Release/node-v43-linux-arm.tar.gz --hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.6.1/Release/ --hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.6.1/Release/node-v43-linux-arm.tar.gz' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:76:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:169:7)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:984:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1057:5)
node-pre-gyp ERR! System Linux 3.13.0-40-generic
node-pre-gyp ERR! command "/usr/bin/node" "/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /node_modules/serialport
node-pre-gyp ERR! node -v v1.4.3
node-pre-gyp ERR! node-pre-gyp -v v0.5.19
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/iojs rebuild --name=serialport --configuration=Release --module_name=serialport --version=1.6.1 --major=1 --minor=6 --patch=1 --runtime=node --node_abi=node-v43 --platform=linux --target_platform=linux --arch=arm --target_arch=arm --module_main=./serialport --host=https://node-serialport.s3.amazonaws.com/ --module_path=/node_modules/serialport/build/serialport/v1.6.1/Release/node-v43-linux-arm --remote_path=./serialport/v1.6.1/Release/ --package_name=node-v43-linux-arm.tar.gz --staged_tarball=build/stage/serialport/v1.6.1/Release/node-v43-linux-arm.tar.gz --hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.6.1/Release/ --hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.6.1/Release/node-v43-linux-arm.tar.gz' (1)
npm ERR! Linux 3.13.0-40-generic
npm ERR! argv "/usr/bin/iojs" "/usr/bin/npm" "install" "serialport"
npm ERR! node v1.4.3
npm ERR! npm  v2.6.1
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.6.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@1.6.1 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /npm-debug.log
root@73f15e99a1de:/# 
```

I've been able to install other packages, some like onoff which also use node-gyp, so i'm not sure what the issue might be:
```
root@73f15e99a1de:/# npm ls --depth=0
/
├── bluebird@2.9.13
├── lodash@3.3.1
├── node-uuid@1.4.3
├── onoff@1.0.2
├── request@2.53.0
├── request-debug@0.1.1
├── request-promise@0.4.0
├── stompjs@2.3.3
├── winston@0.9.0
└── ws@0.7.1
```

techniq, 2015-03-12:
I upgraded to io.js 1.5.0 with npm 2.7.0, which [adds support](https://github.com/npm/npm/blob/master/CHANGELOG.md#a-small-feature-with-big-implications) for `--node-gyp=<PATH_TO_NODE_GYP>`.  I then installed [pangyp](https://github.com/rvagg/pangyp) and tried to install `serialport` using `npm install --node-gyp=node_modules/pangyp/bin/node-gyp.js serialport`, but it failed the same way.  I then saw this [pull request](https://github.com/mapbox/node-pre-gyp/pull/151) for `node-pre-gyp`, which adds support for pangyp (environment variable via the cli argument).

Once that PR is merged, I'll try again.  I'm not sure how bundledDependenies work as serialport has  `node-pre-gyp` as one, but serialport might need to sync up with the changes to node-pre-gyp once the PR is merged.

voodootikigod, 2015-03-12:
Just to clarify, the reason it failed was due to "collect2: fatal error: vfork: Cannot allocate memory
compilation terminated." The reason others might have worked/succeed would be that compilation has a lot to do with the volume of code being built, I haven't looked at on/off. If node-pre-gyp has an update, nudge here -- would be glad to update to the latest. 

Thanks

techniq, 2015-03-12:
Hmm, I saw that but was also having issues building directly on my BBB (just with io.js, not Node.js) so I thought it was directly related to the VM.

I increased the memory from the default (1GB?) to 4GB and was able to `npm install serialport` successfully.  Before I close this, I'm going to do a full project build with all my packages just incase before I close this issue.

techniq, 2015-03-13:
I just tried a full project building on my VM with 4GB and had the following error for serialport

```
> serialport@1.2.5 install /app/node_modules/bonescript/node_modules/serialport
> node-gyp rebuild

child_process: customFds option is deprecated, use stdio instead.
make: Entering directory '/app/node_modules/bonescript/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.cpp:3:0:
../src/serialport.h:30:38: error: 'Arguments' in namespace 'v8' does not name a type
 v8::Handle<v8::Value> List(const v8::Arguments& args);
                                      ^
../src/serialport.h:31:15: error: variable or field 'EIO_List' declared void
 void EIO_List(uv_work_t* req);
               ^
../src/serialport.h:31:15: error: 'uv_work_t' was not declared in this scope
../src/serialport.h:31:26: error: 'req' was not declared in this scope
 void EIO_List(uv_work_t* req);
                          ^
../src/serialport.h:32:20: error: variable or field 'EIO_AfterList' declared void
 void EIO_AfterList(uv_work_t* req);
                    ^
../src/serialport.h:32:20: error: 'uv_work_t' was not declared in this scope
../src/serialport.h:32:31: error: 'req' was not declared in this scope
 void EIO_AfterList(uv_work_t* req);
                               ^
../src/serialport.h:34:38: error: 'Arguments' in namespace 'v8' does not name a type
 v8::Handle<v8::Value> Open(const v8::Arguments& args);
                                      ^
../src/serialport.h:35:15: error: variable or field 'EIO_Open' declared void
 void EIO_Open(uv_work_t* req);
               ^
../src/serialport.h:35:15: error: 'uv_work_t' was not declared in this scope
../src/serialport.h:35:26: error: 'req' was not declared in this scope
 void EIO_Open(uv_work_t* req);
                          ^
../src/serialport.h:36:20: error: variable or field 'EIO_AfterOpen' declared void
 void EIO_AfterOpen(uv_work_t* req);
                    ^
../src/serialport.h:36:20: error: 'uv_work_t' was not declared in this scope
../src/serialport.h:36:31: error: 'req' was not declared in this scope
 void EIO_AfterOpen(uv_work_t* req);
                               ^
In file included from ../src/serialport.cpp:3:0:
../src/serialport.h:39:39: error: 'Arguments' in namespace 'v8' does not name a type
 v8::Handle<v8::Value> Write(const v8::Arguments& args);
                                       ^
../src/serialport.h:40:16: error: variable or field 'EIO_Write' declared void
 void EIO_Write(uv_work_t* req);
                ^
../src/serialport.h:40:16: error: 'uv_work_t' was not declared in this scope
../src/serialport.h:40:27: error: 'req' was not declared in this scope
 void EIO_Write(uv_work_t* req);
                           ^
../src/serialport.h:41:21: error: variable or field 'EIO_AfterWrite' declared void
 void EIO_AfterWrite(uv_work_t* req);
                     ^
../src/serialport.h:41:21: error: 'uv_work_t' was not declared in this scope
../src/serialport.h:41:32: error: 'req' was not declared in this scope
 void EIO_AfterWrite(uv_work_t* req);
                                ^
../src/serialport.h:43:39: error: 'Arguments' in namespace 'v8' does not name a type
 v8::Handle<v8::Value> Close(const v8::Arguments& args);
                                       ^
../src/serialport.h:44:16: error: variable or field 'EIO_Close' declared void
 void EIO_Close(uv_work_t* req);
                ^
../src/serialport.h:44:16: error: 'uv_work_t' was not declared in this scope
../src/serialport.h:44:27: error: 'req' was not declared in this scope
[... 311 lines cut ...]
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/iojs rebuild --name=serialport --configuration=Release --module_name=serialport --version=1.4.6 --major=1 --minor=4 --patch=6 --runtime=node --node_abi=node-v43 --platform=linux --target_platform=linux --arch=arm --target_arch=arm --module_main=./serialport --host=https://node-serialport.s3.amazonaws.com/ --module_path=/app/node_modules/pn532/node_modules/serialport/build/serialport/v1.4.6/Release/node-v43-linux-arm --remote_path=./serialport/v1.4.6/Release/ --package_name=node-v43-linux-arm.tar.gz --staged_tarball=build/stage/serialport/v1.4.6/Release/node-v43-linux-arm.tar.gz --hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.4.6/Release/ --hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.4.6/Release/node-v43-linux-arm.tar.gz' (1)
npm WARN optional dep failed, continuing serialport@1.4.6
npm ERR! Linux 3.13.0-40-generic
npm ERR! argv "/usr/bin/iojs" "/usr/bin/npm" "install"
npm ERR! node v1.5.0
npm ERR! npm  v2.7.0
npm ERR! code ELIFECYCLE

npm ERR! epoll@0.0.7 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the epoll@0.0.7 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the epoll package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls epoll
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /app/npm-debug.log
2015/03/13 15:41:30 The command [/bin/sh -c npm install -g gulp 6to5 && npm install] returned a non-zero code: 1
```

techniq, 2015-03-16:
Turns out the issue with building my project was I have a dependency on `serialport = ^1.4.8` but also `bonescript`, which has an [optionalDependency](https://github.com/jadonk/bonescript/blob/master/package.json#L37) on `serialport = 1.2.5`.  I removed the dependency to bonescript (as I've moved on to using other libraries such as [onoff](https://github.com/fivdi/onoff) for gpio) and the build issue went away.

Sorry for the noise and thanks for the nudge about the memory issue.

closed 2015-03-16 by techniq