serialport/node-serialport #479: PR 476 duplicates BREAK functionality from PR 470

jacobrosenthal, 2015-03-10:
Recent PR https://github.com/voodootikigod/node-serialport/pull/476 duplicates existing 
break functionality added https://github.com/voodootikigod/node-serialport/pull/470 and released as 1.6.1

Obviously just an oversight as they landed so close to eachother.

@jerome-42 is there can you test the existing cross platform functionality that now exists in upstream 1.6.1 functionality and ensure that it covers your needs?

voodootikigod, 2015-03-15:
How do I back this out or finalize a proper conclusion @jacobrosenthal ?

jacobrosenthal, 2015-03-15:
My suggestion is just that, back out this new commit as it uses a whole new
command path. Rebase or just patch it back out is up to you.

jerome-42, 2015-03-15:
Sorry for my late response.

I tried set({ brk: true }) with no luck, Cisco router doesn't see break signal.

I think Cisco router is a bit of old school console controller which doesn't see normal break signal. Controller needs a loooooong signal break.

This explain the 1000 parameter, which will be setable from JS code, as delay varies with console speed. 1000 is fine for Cisco router at 9600 baud speed.

jacobrosenthal, 2015-03-15:
Brk true is infinite break signal if you dont then manually brk false. So
length cant be your issue. Check the example:
https://github.com/voodootikigod/node-serialport/blob/master/examples/break.js

jerome-42, 2015-03-16:
It works with set({ brk: true }).
Then my code is useless. You can revert the merge.

jacobrosenthal, 2015-03-16:
Not at all useless I just happened to beat you to the punch this time :)

jerome-42, 2015-03-16:
No problem :)

jacobrosenthal, 2015-03-16:
@jerome-42 You could actually do me a huge favor. Im petitioning to get Break functionality into chrome as its not currently supported. If you or any of your other Cisco friends could +1 this ticket that might move it forward:
https://code.google.com/p/chromium/issues/detail?id=461169&q=break&colspec=ID%20Pri%20M%20Week%20ReleaseBlock%20Cr%20Status%20Owner%20Summary%20OS%20Modified

jerome-42, 2015-03-16:
Done :)

PR #490 (2015-03-17): Revert "Add BREAK command, useful for CISCO router to break into rommon"
This reverts commit f37d1e690d4714c0ed77ffa57a547f49b619f06f. Fixes #479
closed 2015-03-17 by voodootikigod