serialport/node-serialport #480: serialport is not compatible with node-webkit (nw.js) v0.12.0

roeltz, 2015-03-10:
When building with node-pre-gyp (`node-pre-gyp build --runtime=node-webkit --target=0.12.0`), compiler logs some errors and build is not able to succeed.

	E:\...\node_modules\serialport\node_modules\nan\nan.h(481): error C2039: 'ExternalAsciiStringResource' : is not a member of 'v8::String' (..\src\serialport_win.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	          C:\Users\User\.nw-gyp\0.12.0\deps\v8\include\v8.h(1885) : see declaration of 'v8::String'
	E:\...\node_modules\serialport\node_modules\nan\nan.h(481): error C2065: 'ExternalAsciiStringResource' : undeclared identifier (..\src\serialport_win.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(481): error C2065: 'resource' : undeclared identifier (..\src\serialport_win.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(481): error C2448: 'NanNew' : function-style initializer appears to be a function definition (..\src\serialport_win.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2103): error C2039: 'IsExternalAscii' : is not a member of 'v8::String' (..\src\serialport_win.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	          C:\Users\User\.nw-gyp\0.12.0\deps\v8\include\v8.h(1885) : see declaration of 'v8::String'
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2104): error C2039: 'ExternalAsciiStringResource' : is not a member of 'v8::String' (..\src\serialport_win.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	          C:\Users\User\.nw-gyp\0.12.0\deps\v8\include\v8.h(1885) : see declaration of 'v8::String'
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2104): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int (..\src\serialport_win.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2104): error C2143: syntax error : missing ';' before '*' (..\src\serialport_win.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2104): error C2065: 'ext' : undeclared identifier (..\src\serialport_win.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2105): error C2065: 'ext' : undeclared identifier (..\src\serialport_win.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2105): error C2039: 'GetExternalAsciiStringResource' : is not a member of 'v8::String' (..\src\serialport_win.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	          C:\Users\User\.nw-gyp\0.12.0\deps\v8\include\v8.h(1885) : see declaration of 'v8::String'
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2106): error C2065: 'ext' : undeclared identifier (..\src\serialport_win.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2106): error C2227: left of '->data' must point to class/struct/union/generic type (..\src\serialport_win.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	          type is ''unknown-type''
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2107): error C2065: 'ext' : undeclared identifier (..\src\serialport_win.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2107): error C2227: left of '->length' must point to class/struct/union/generic type (..\src\serialport_win.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	          type is ''unknown-type''
	E:\...\node_modules\serialport\node_modules\nan\nan.h(481): error C2039: 'ExternalAsciiStringResource' : is not a member of 'v8::String' (..\src\serialport.cpp)[E:\...\node_modules\serialport\build\serialport.vcxproj]
	          C:\Users\User\.nw-gyp\0.12.0\deps\v8\include\v8.h(1885) : see declaration of 'v8::String'
	E:\...\node_modules\serialport\node_modules\nan\nan.h(481): error C2065: 'ExternalAsciiStringResource' : undeclared identifier (..\src\serialport.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(481): error C2065: 'resource' : undeclared identifier (..\src\serialport.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(481): error C2448: 'NanNew' : function-style initializer appears to be a function definition (..\src\serialport.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2103): error C2039: 'IsExternalAscii' : is not a member of 'v8::String' (..\src\serialport.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	          C:\Users\User\.nw-gyp\0.12.0\deps\v8\include\v8.h(1885) : see declaration of 'v8::String'
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2104): error C2039: 'ExternalAsciiStringResource' : is not a member of 'v8::String' (..\src\serialport.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	          C:\Users\User\.nw-gyp\0.12.0\deps\v8\include\v8.h(1885) : see declaration of 'v8::String'
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2104): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int (..\src\serialport.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2104): error C2143: syntax error : missing ';' before '*' (..\src\serialport.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2104): error C2065: 'ext' : undeclared identifier (..\src\serialport.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2105): error C2065: 'ext' : undeclared identifier (..\src\serialport.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2105): error C2039: 'GetExternalAsciiStringResource' : is not a member of 'v8::String' (..\src\serialport.c
	pp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	          C:\Users\User\.nw-gyp\0.12.0\deps\v8\include\v8.h(1885) : see declaration of 'v8::String'
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2106): error C2065: 'ext' : undeclared identifier (..\src\serialport.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2106): error C2227: left of '->data' must point to class/struct/union/generic type (..\src\serialport.cpp)
	[E:\...\node_modules\serialport\build\serialport.vcxproj]
	          type is ''unknown-type''
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2107): error C2065: 'ext' : undeclared identifier (..\src\serialport.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	E:\...\node_modules\serialport\node_modules\nan\nan.h(2107): error C2227: left of '->length' must point to class/struct/union/generic type (..\src\serialport.cpp) [E:\...\node_modules\serialport\build\serialport.vcxproj]
	          type is ''unknown-type''

How could I get to successfully build serialport for node-webkit 0.12.0?

voodootikigod, 2015-03-15:
This normally has to do with a conflict of NAN between the two.

blairconner, 2015-04-20:
I have a similar problem using serialport with node-webkit v0.12.0 which I believe is related to the OP. While I don't receive the same error as above, when I include the serialport module I receive the following error:

Error: Cannot find module 'c:\GIT_Repositories\TI\OneUI\oneui-app-src\node_modules\SspEvmNodeJS\node_modules\serialport\build\serialport\v1.6.3\Release\node-v43-win32-x64\serialport.node'
    at Function.Module._resolveFilename (module.js:327:15)
    at Function.Module._load (module.js:264:25)
    at Module.require (module.js:356:17)
    at require (module.js:375:17)
    at Object.<anonymous> (c:\GIT_Repositories\TI\OneUI\oneui-app-src\node_modules\SspEvmNodeJS\node_modules\serialport\serialport.js:14:25)
    at Module._compile (module.js:451:26)
    at Object.Module._extensions..js (module.js:469:10)
    at Module.load (module.js:346:32)
    at Function.Module._load (module.js:301:12)
    at Module.require (module.js:356:17)

blairconner, 2015-04-20:
Looks like I've come up with a work around of sorts, although it isn't the work around that leaves you feeling warm and fuzzy inside.

1.	From “node_modules\serialport” I ran the following commands…
     a.	node-pre-gyp configure --runtime=node-webkit --target=0.12.0 -target-platform=win64
     b.	node-pre-gyp build --runtime=node-webkit --target=0.12.0 -target-platform=win64
2.	Went to “\node_modules\serialport\build\serialport\v1.6.3\Release”
     a.	Copied folder “node-webkit-v0.12.0-win32-x64”
     b.	Renamed copied folder to “node-v43-win32-x64”

After that the serialport module has been running without issues. I assume I need to modify the node-pre-gyp commands to avoid my workaround, but am unsure what needs to change.

voodootikigod, 2015-05-09:
This is due to the way nw.js/node-webkit changes the Node-ABI. This is not an issue with serialport, review this for more information: https://github.com/nwjs/nw.js/wiki/Using-Node-modules#3rd-party-modules-with-cc-addons

Closing

closed 2015-05-09 by voodootikigod