serialport/node-serialport #481: Cannot read property 'forEach' of undefined when running   ports.forEach(function(port) {

xuqianli, 2015-03-11:
Hi,

I am working on a project that connects raspberry pi and arduino together. I seems to have issue with  ports.forEach(function(port) {...} when my arduino is not plugged in with my raspberry pi. It was programmed to prompted a line in console instead to say arduino not connected.

Thank you very much 

**Code**:
```javascript
findArduinoOnPi = function () {
  var port;

  serialport.list(function (err, ports) {

    ports.forEach(function(obj) {

      if (obj.hasOwnProperty('pnpId')){
        // FTDI captures the duemilanove //
        // Arduino captures the leonardo //
        if (obj.pnpId.search('FTDI') != -1 || obj.pnpId.search('Arduino') != -1) {
          port = obj.comName;
        }
      }
    });

    if (port){

      portConnect(port);

    } else { 

      console.log('* failed to find arduino : please check your connections *');
    }  
  });
}
```
**Error**:
```
/home/pi/workSpace/0xb0t/modules/serial_communication.js:62
    ports.forEach(function(obj) {
         ^
TypeError: Cannot read property 'forEach' of undefined
    at /home/pi/workSpace/0xb0t/modules/serial_communication.js:62:10
    at /home/pi/workSpace/0xb0t/node_modules/serialport/serialport.js:523:11
    at FSReqWrap.oncomplete (fs.js:99:15)

npm ERR! Linux 3.18.7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
npm ERR! node v0.12.0
npm ERR! npm  v2.5.1
npm ERR! code ELIFECYCLE
npm ERR! 0xb0t@0.0.0 start: `node ./bin/www`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the 0xb0t@0.0.0 start script 'node ./bin/www'.
npm ERR! This is most likely a problem with the 0xb0t package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node ./bin/www
npm ERR! You can get their info via:
npm ERR!     npm owner ls 0xb0t
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/workSpace/0xb0t/npm-debug.log
```

voodootikigod, 2015-03-15:
can you console.log(err) to see if the system is throwing an error?

elafargue, 2015-03-17:
I have the same issue: serialport.list is currently broken on recent Linux versions - node-serialport tries to get the list of serial ports from /dev/serial/by-id and /dev/serial does not exist on a lot of recent Linux systems - same problem on Ubuntu 14.01 on BeagleBone Black.

llibdude, 2015-07-17:
Are there any plans to support Ubuntu 14.x in the near future?  I am running into the same issue.  I am using node-serialport 1.7.4.

As elafargue already mentioned, when I console the error I get:
code: "ENOENT"
errno: -2
message: "ENOENT, readdir '/dev/serial/by-id'"
path: "/dev/serial/by-id"

lfk, 2015-09-09:
Bumping this. `Linux 4.1.6-1-ARCH#1 SMP PREEMPT Mon Aug 17 08:52:28 CEST 2015 x86_64 GNU/Linux`

ArekSredzki, 2015-10-15:
This is also true for Raspian.

```javascript
{
[Error: ENOENT, scandir '/dev/serial/by-id']
errno: -2,
code: 'ENOENT',
path: '/dev/serial/by-id'
}
```

jacobrosenthal, 2015-10-15:
For what version of raspbian, nodejs, serialport

ArekSredzki, 2015-10-15:
RASPBIAN JESSIE
node v0.12.6
serialport 2.0.1

jacobrosenthal, 2015-10-15:
Confirmed on 12 and 4.. Oddly enough it doesnt fail on node 10...

ArekSredzki, 2015-10-15:
interesting

jacobrosenthal, 2015-10-24:
were tracking this https://github.com/voodootikigod/node-serialport/issues/610

closed 2015-10-24 by jacobrosenthal