serialport/node-serialport #483: Cannot install on Windows: unresolved external symbol "void __cdecl EIO_Break(struct uv_work_s *)"

Domiii, 2015-03-15:
I have: Windows 7, Python 2.7.6, Node 0.10.32 and Visual Studio 2013.

When installing `duino`, the `serialport` build fails with the following message:

    serialport.obj : error LNK2001: unresolved external symbol "void __cdecl EIO_Break(struct uv_work_s *)"

Apparently, `EIO_Break` is defined in [serialport_unix.cpp](https://github.com/voodootikigod/node-serialport/blob/17ea1fc0e8c42908fceed90de94c0e5cbe4d76bb/src/serialport_unix.cpp#L745).

Does that mean that I need Cygwin for this to work or something? It does not mention that in the [installation instructions](https://github.com/voodootikigod/node-serialport#windows) though...

voodootikigod, 2015-03-15:
You shouldn't I don't have my surface with me, but will check it when I get
to work.

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

Domiii, 2015-03-15:
Apparently, [it was added 9 days ago](https://github.com/voodootikigod/node-serialport/commit/f37d1e690d4714c0ed77ffa57a547f49b619f06f#diff-4f97e03868bd8b1d8ea209b7a03863abR592).

Someone forgot about Windows in that commit :/

Domiii, 2015-03-15:
Also adding a Windows build to the [travis build matrix](https://travis-ci.org/voodootikigod/node-serialport) would probably be a good idea! :)

v12, 2015-03-16:
I can confirm that I am experiencing the same issue.

Domiii, 2015-03-16:
In order to support this function on Windows, it all boils down to `tcsendbreak`, whose Windows equivalent seems to be discussed in [this SO thread](http://stackoverflow.com/questions/4585661/sending-serial-break-using-windows-xp-api), which in turn refers to [this article on serial programming on Windows](http://www.codeguru.com/cpp/i-n/network/serialcommunications/article.php/c5425/Serial-Communication-in-Windows.htm).

jacobrosenthal, 2015-03-16:
This functionality is as you see minimally implemented and as it turns out also duplicate and will be removed shortly. See #479 In the mean time just lock down to a version below 1.6.2

Domiii, 2015-03-17:
Awesome, thanks!

guywithcrookedface, 2015-03-17:
I am also having this issue, reverting to 1.6.1 fixed the issue for now.

DaAwesomeP, 2015-03-18:
Is this fixed in 1.6.3?

voodootikigod, 2015-03-18:
Should be -- can you test?

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

DaAwesomeP, 2015-03-18:
Removing it and installing it again works fine. However, `npm update` failed.

closed 2015-10-11 by jacobrosenthal