serialport/node-serialport #489: Unable to install serialport v1.6.2 on raspberry.

ipqate, 2015-03-17:
Cannot install on raspberry pi model B+ running raspbian from NOOBS, node v0.10.12 and npm v1.2.32

npm install serialport@1.4.2 ---> WORKS!!! (But we need some features developed in the last release...)

------------------------------------------------------------------------------------------------------------
pi@EMBED1 ~/testserial $ npm install serialport
npm http GET https://registry.npmjs.org/serialport
npm http 304 https://registry.npmjs.org/serialport
npm http GET https://registry.npmjs.org/async/0.9.0
npm http GET https://registry.npmjs.org/bindings/1.2.1
npm http GET https://registry.npmjs.org/debug
npm http GET https://registry.npmjs.org/nan/1.7.0
npm http GET https://registry.npmjs.org/optimist
npm http GET https://registry.npmjs.org/sf/0.1.7
npm http 304 https://registry.npmjs.org/async/0.9.0
npm http 304 https://registry.npmjs.org/debug
npm http 304 https://registry.npmjs.org/bindings/1.2.1
npm ERR! error rolling back Error: EACCES, unlink '/home/pi/testserial/node_modules/.bin/serialportlist'
npm ERR! error rolling back  serialport@1.6.2 { [Error: EACCES, unlink '/home/pi/testserial/node_modules/.bin/serialportlist']
npm ERR! error rolling back   errno: 3,
npm ERR! error rolling back   code: 'EACCES',
npm ERR! error rolling back   path: '/home/pi/testserial/node_modules/.bin/serialportlist' }
npm ERR! Error: No compatible version found: debug@'^2.1.1'
npm ERR! Valid install targets:
npm ERR! ["0.0.1","0.1.0","0.2.0","0.3.0","0.4.0","0.4.1","0.5.0","0.6.0","0.7.0","0.7.1","0.7.2","0.7.3","0.7.4","0.8.0","0.8.1","1.0.0","1.0.1","1.0.2","1.0.3","1.0.4","2.0.0","2.1.0","2.1.1","2.1.2","2.1.3"]
npm ERR!     at installTargetsError (/opt/node/lib/node_modules/npm/lib/cache.js:719:10)
npm ERR!     at /opt/node/lib/node_modules/npm/lib/cache.js:641:10
npm ERR!     at saved (/opt/node/lib/node_modules/npm/node_modules/npm-registry-client/lib/get.js:138:7)
npm ERR!     at Object.oncomplete (fs.js:107:15)
npm ERR! If you need help, you may report this log at:
npm ERR!     <http://github.com/isaacs/npm/issues>
npm ERR! or email it to:
npm ERR!     <npm-@googlegroups.com>

npm ERR! System Linux 3.18.7+
npm ERR! command "/opt/node/bin/node" "/opt/node/bin/npm" "install" "serialport"
npm ERR! cwd /home/pi/testserial
npm ERR! node -v v0.10.12
npm ERR! npm -v 1.2.32
npm http 304 https://registry.npmjs.org/optimist
npm http 304 https://registry.npmjs.org/sf/0.1.7
npm http 304 https://registry.npmjs.org/nan/1.7.0
npm WARN package.json nan@1.7.0 No readme data.
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/testserial/npm-debug.log
npm ERR! not ok code 0
 

Any help????????????????????
Thanks in advance

jacobrosenthal, 2015-03-17:
That seems odd. It says it cant find a package match for debug@'^2.1.1 (greater than or equal to 2.1.1 but less than 3.0) and then lists three options that should have match that 2.1.1,2.1.2,2.1.3

Can you successfully install one of those seperately, like npm i debug@2.1.1

marcuswinkler, 2015-03-17:
The problem seems to be the ^ (caret) versíon constraint for the debug package. As far as I can tell, support for this constraint type was added with node v0.10.16.

I can confirm that serialport installs successfully on a Raspberry Pi running Raspbian with v0.10.16.

ipqate, 2015-03-18:
Hello, node updated to v0.10.16.

pi@EMBED1 ~/testserial $ npm install serialport
npm http GET https://registry.npmjs.org/serialport
npm http 200 https://registry.npmjs.org/serialport
npm http GET https://registry.npmjs.org/serialport/-/serialport-1.6.3.tgz
npm http 200 https://registry.npmjs.org/serialport/-/serialport-1.6.3.tgz
npm http GET https://registry.npmjs.org/nan/1.7.0
npm http GET https://registry.npmjs.org/optimist
npm http GET https://registry.npmjs.org/sf/0.1.7
npm http GET https://registry.npmjs.org/async/0.9.0
npm http GET https://registry.npmjs.org/bindings/1.2.1
npm http GET https://registry.npmjs.org/debug
npm http 200 https://registry.npmjs.org/nan/1.7.0
npm http 200 https://registry.npmjs.org/optimist
npm http 200 https://registry.npmjs.org/debug
npm http 200 https://registry.npmjs.org/sf/0.1.7
npm http 200 https://registry.npmjs.org/async/0.9.0
npm http GET https://registry.npmjs.org/optimist/-/optimist-0.6.1.tgz
npm http GET https://registry.npmjs.org/nan/-/nan-1.7.0.tgz
npm http GET https://registry.npmjs.org/debug/-/debug-2.1.3.tgz
npm http 200 https://registry.npmjs.org/bindings/1.2.1
npm http GET https://registry.npmjs.org/sf/-/sf-0.1.7.tgz
npm http GET https://registry.npmjs.org/async/-/async-0.9.0.tgz
npm http GET https://registry.npmjs.org/bindings/-/bindings-1.2.1.tgz
npm http 200 https://registry.npmjs.org/optimist/-/optimist-0.6.1.tgz
npm http 200 https://registry.npmjs.org/bindings/-/bindings-1.2.1.tgz
npm http 200 https://registry.npmjs.org/nan/-/nan-1.7.0.tgz
npm http 200 https://registry.npmjs.org/debug/-/debug-2.1.3.tgz
npm http 200 https://registry.npmjs.org/sf/-/sf-0.1.7.tgz
npm http 200 https://registry.npmjs.org/async/-/async-0.9.0.tgz
npm http GET https://registry.npmjs.org/ms/0.7.0
npm http 200 https://registry.npmjs.org/ms/0.7.0
npm http GET https://registry.npmjs.org/wordwrap
npm http GET https://registry.npmjs.org/minimist
npm http GET https://registry.npmjs.org/ms/-/ms-0.7.0.tgz
npm http 200 https://registry.npmjs.org/wordwrap
npm http GET https://registry.npmjs.org/wordwrap/-/wordwrap-0.0.2.tgz
npm http 200 https://registry.npmjs.org/minimist
npm http GET https://registry.npmjs.org/minimist/-/minimist-0.0.10.tgz
npm http 200 https://registry.npmjs.org/ms/-/ms-0.7.0.tgz
npm http 200 https://registry.npmjs.org/wordwrap/-/wordwrap-0.0.2.tgz
npm http 200 https://registry.npmjs.org/minimist/-/minimist-0.0.10.tgz
npm ERR! error rolling back Error: EACCES, unlink '/home/pi/testserial/node_modules/.bin/serialportlist'
npm ERR! error rolling back  serialport@1.6.3 { [Error: EACCES, unlink '/home/pi/testserial/node_modules/.bin/serialportlist']
npm ERR! error rolling back   errno: 3,
npm ERR! error rolling back   code: 'EACCES',
npm ERR! error rolling back   path: '/home/pi/testserial/node_modules/.bin/serialportlist' }
npm ERR! Error: EACCES, unlink '/home/pi/testserial/node_modules/.bin/serialportlist'
npm ERR!  { [Error: EACCES, unlink '/home/pi/testserial/node_modules/.bin/serialportlist']
npm ERR!   errno: 3,
npm ERR!   code: 'EACCES',
npm ERR!   path: '/home/pi/testserial/node_modules/.bin/serialportlist' }
npm ERR!
npm ERR! Please try running this command again as root/Administrator.

npm ERR! System Linux 3.18.7+
npm ERR! command "/opt/node/bin/node" "/opt/node/bin/npm" "install" "serialport"
npm ERR! cwd /home/pi/testserial
npm ERR! node -v v0.10.16
npm ERR! npm -v 1.3.8
npm ERR! path /home/pi/testserial/node_modules/.bin/serialportlist
npm ERR! code EACCES
npm ERR! errno 3
npm ERR! stack Error: EACCES, unlink '/home/pi/testserial/node_modules/.bin/serialportlist'
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/testserial/npm-debug.log
npm ERR! not ok code 0

However, using: npm install -g serialport --------------->SUCCEED

Then, launching test app: > node test.js -----------------------------> FAILED

SOLUTION: Files moved from /opt/node/lib/node_modules to the local app (under node_modules) ----------- > SUCCESS

GlennGeenen, 2015-05-19:
I also get npm ERR! Error: No compatible version found: debug@'^2.1.1'

on node v0.10.3 installing serialport 1.7.0

closed 2015-09-14 by jacobrosenthal