serialport/node-serialport #492: Tests fail on node 12.

edgarsilva, 2015-03-25:
Tests are failing on node 12 which is now the stable version, this prevents us from adding it to the CI tools to create pre-compiled versions for that version of node.

```bash
Options: reporter="spec"

  parsers
    #raw
      ✓ emits data exactly as it's written 
    #readline
      ✓ emits data events split on a delimiter 
    #byteLength
      ✓ emits data events every 8 bytes 

  SerialPort
    Constructor
      ✓ opens the port immediately 
      ✓ emits the open event 
      ✓ emits an error on the factory when erroring without a callback 
      ✓ emits an error on the serialport when explicit error handler present 
      ✓ errors with invalid databits 
      ✓ errors with invalid stopbits 
      ✓ errors with invalid parity 
      ✓ errors with invalid flow control 
      ✓ allows optional options 
    Functions
      ✓ write errors when serialport not open 
      ✓ close errors when serialport not open 
      ✓ flush errors when serialport not open 
      ✓ set errors when serialport not open 
      ✓ drain errors when serialport not open 
    reading data
      ✓ emits data events by default 
      ✓ calls the dataCallback if set 
    #open
      ✓ passes the port to the bindings 
      ✓ calls back an error when opening an invalid port 
      ✓ emits data after being reopened 
    close
      ✓ fires a close event when it's closed 
      ✓ fires a close event after being reopened 
      ✓ emits a close event 
    disconnect
      ✓ fires a disconnect event 
      ✓ emits a disconnect event 

  SerialPort
    Initialization
      1) Throws an error in callback when trying to open an invalid port

  27 passing (32ms)
  1 failing

  1) SerialPort Initialization Throws an error in callback when trying to open an invalid port:
     Error: Module did not self-register.
      at Error (native)
      at Module.load (module.js:355:32)
      at Function.Module._load (module.js:310:12)
      at Module.require (module.js:365:17)
      at require (module.js:384:17)
      at Object.<anonymous> (/home/edgar/workspace/node-serialport/serialport.js:14:25)
      at Module._compile (module.js:460:26)
      at Object.Module._extensions..js (module.js:478:10)
      at Module.load (module.js:355:32)
      at Function.Module._load (module.js:310:12)

Warning: Task "mochaTest:test" failed. Use --force to continue.

Aborted due to warnings.
npm ERR! Test failed.  See above for more details.
```

edgarsilva, 2015-03-25:
This seems to be an issue with mocha tests, if we run the tests individually with node 12 they all work. If we run all of them either with `mocha test`, `npm test` or `grunt test`, then `test/serialport-c.js` fails.

deadprogram, 2015-03-28:
This issue results in no automatic binary builds for node 12.x so it is probably worth escalating. Jus' sayin'...

deadprogram, 2015-04-01:
Ping?

deadprogram, 2015-04-02:
@reconbot can you please help with this? You surely know the test code better than anyone! :trophy: :

jacobrosenthal, 2015-04-11:
Take out the if statement and it blows up on requiring node-serialport line 
var SerialPortBinding = require(binding_path);

jacobrosenthal, 2015-04-11:
if/else was originally implemented by @voodootikigod https://github.com/voodootikigod/node-serialport/commit/f37144ca12f334fb7c61cdbe3c31326f478e5fa8 maybe he can help

reconbot, 2016-04-03:
@edgarsilva this is no longer an issue, I bet it was due to requiring the bindings in two different contexts, a sandboxed one and a regular one. We no longer require the bindings in our sandboxed tests, and since then we could take out the if/else.

Let me know if you'd like any more followup.

closed 2016-04-03 by reconbot