serialport/node-serialport #494: npm shrinkwrap fails

wildeyes, 2015-03-30:
running npm i serialport && npm shrinkwrap fails with the following output (I want to shrinkwrap for meteor)
0 info it worked if it ends with ok
1 verbose cli [ 'C:\\Program Files\\nodejs\\\\node.exe',
1 verbose cli   'C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js',
1 verbose cli   'shrinkwrap' ]
2 info using npm@2.5.1
3 info using node@v0.12.1
4 verbose node symlink C:\Program Files\nodejs\\node.exe
5 verbose stack Error: Problems were encountered
5 verbose stack Please correct and try again.
5 verbose stack invalid: node-pre-gyp@0.5.19 C:\Users\Administrator\User\test\node_modules\serialport\node_modules\node-pre-gyp
5 verbose stack invalid: readable-stream@1.0.27-1 C:\Users\Administrator\User\test\node_modules\serialport\node_modules\node-pre-gyp\node_modules\tar-pack\node_modules\readable-stream
5 verbose stack invalid: string_decoder@0.10.25-1 C:\Users\Administrator\User\test\node_modules\serialport\node_modules\node-pre-gyp\node_modules\tar-pack\node_modules\readable-stream\node_modules\string_decoder
5 verbose stack     at shrinkwrap_ (C:\Program Files\nodejs\node_modules\npm\lib\shrinkwrap.js:31:15)
5 verbose stack     at C:\Program Files\nodejs\node_modules\npm\lib\shrinkwrap.js:25:5
5 verbose stack     at C:\Program Files\nodejs\node_modules\npm\lib\ls.js:46:30
5 verbose stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\read-installed\read-installed.js:138:5
5 verbose stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\read-installed\read-installed.js:251:14
5 verbose stack     at cb (C:\Program Files\nodejs\node_modules\npm\node_modules\slide\lib\async-map.js:47:24)
5 verbose stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\read-installed\read-installed.js:251:14
5 verbose stack     at cb (C:\Program Files\nodejs\node_modules\npm\node_modules\slide\lib\async-map.js:47:24)
5 verbose stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\read-installed\read-installed.js:251:14
5 verbose stack     at cb (C:\Program Files\nodejs\node_modules\npm\node_modules\slide\lib\async-map.js:47:24)
6 verbose cwd C:\Users\Administrator\User\test
7 error Windows_NT 6.1.7600
8 error argv "C:\\Program Files\\nodejs\\\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "shrinkwrap"
9 error node v0.12.1
10 error npm  v2.5.1
11 error Problems were encountered
11 error Please correct and try again.
11 error invalid: node-pre-gyp@0.5.19 C:\Users\Administrator\User\test\node_modules\serialport\node_modules\node-pre-gyp
11 error invalid: readable-stream@1.0.27-1 C:\Users\Administrator\User\test\node_modules\serialport\node_modules\node-pre-gyp\node_modules\tar-pack\node_modules\readable-stream
11 error invalid: string_decoder@0.10.25-1 C:\Users\Administrator\User\test\node_modules\serialport\node_modules\node-pre-gyp\node_modules\tar-pack\node_modules\readable-stream\node_modules\string_decoder
12 error If you need help, you may report this error at:
12 error     <http://github.com/npm/npm/issues>
13 verbose exit [ 1, true ]

wildeyes, 2015-03-30:
Has to do with npm/npm#6556 and node-pre-gyp with his darn out-dated dependencies. Not related to node-serialport.

wildeyes, 2015-03-31:
This is weird, running `npm init && npm install --save serialport && npm ls` gives me that serialport installed via remote (couldn't figure what that meant, i think it's related), and that node-pre-gyp's version is 0.5.9 although in serialport's package.json it says `0.6.x`.

Why isn't serialport installing the latest node-pre-gyp version it requires?

reconbot, 2016-03-30:
It should be, let us know if this is still an issue.

closed 2016-03-30 by reconbot