serialport/node-serialport #496: Unable to install on mac

kennethlimcp, 2015-04-07:
We have a user with a problem getting `serialport` installed on his Mac OSx 10.10.2.

See: https://community.spark.io/t/tutorial-spark-cli-on-mac-osx-25-july-2014/5225/70 and https://community.spark.io/t/tutorial-spark-cli-on-mac-osx-25-july-2014/5225/85

Can someone help figure out what might be causing the issue? Thanks!

voodootikigod, 2015-04-07:
Appears that node is not in the path

```
env: node: No such file or directory
```

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

kennethlimcp, 2015-04-10:
Thanks! Will check and get back again

kennethlimcp, 2015-04-16:
@voodootikigod , seems like node was in the path `npm ERR! node -v v0.10.29`

zsup, 2015-05-11:
I just ran into this same issue. It seems to have been caused by the fact that my instance of `spark-cli` that I was working with was installed in a Dropbox folder that had spaces and parentheses in the name. When I tried again in a different folder (outside of the Dropbox folder), `npm install` succeeded.

Not sure whether the problem was Dropbox or the path name with spaces and parentheses.

zsup, 2015-05-11:
Here's a gist of the logs from my failed attempts to `npm install serialport`

https://gist.github.com/zsup/f325c09b3ecc205376cf

sarfata, 2015-05-30:
Same problem here. Slightly different message. 

```
gyp verb build dir attempting to create "build" dir: /Users/thomas/work-perso/node-serialport/build
gyp verb build dir "build" dir needed to be created? /Users/thomas/work-perso/node-serialport/build
gyp verb build/config.gypi creating config file
gyp verb build/config.gypi writing out config file: /Users/thomas/work-perso/node-serialport/build/config.gypi
gyp verb config.gypi checking for gypi file: /Users/thomas/work-perso/node-serialport/config.gypi
gyp verb common.gypi checking for gypi file: /Users/thomas/work-perso/node-serialport/common.gypi
gyp verb gyp gyp format was not specified; forcing "make"
gyp info spawn python
gyp info spawn args [ '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/thomas/work-perso/node-serialport/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/thomas/.node-gyp/0.12.3/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/Users/thomas/.node-gyp/0.12.3',
gyp info spawn args   '-Dmodule_root_dir=/Users/thomas/work-perso/node-serialport',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info ok
gyp info it worked if it ends with ok
gyp verb cli [ '/usr/local/bin/node',
gyp verb cli   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js',
gyp verb cli   'build',
gyp verb cli   '--fallback-to-build',
gyp verb cli   '--module=/Users/thomas/work-perso/node-serialport/build/serialport/v1.7.0/Release/node-v14-darwin-x64/serialport.node',
gyp verb cli   '--module_name=serialport',
gyp verb cli   '--module_path=/Users/thomas/work-perso/node-serialport/build/serialport/v1.7.0/Release/node-v14-darwin-x64' ]
gyp info using node-gyp@1.0.3
gyp info using node@0.12.3 | darwin | x64
gyp verb command build []
gyp verb build type Release
gyp verb architecture x64
gyp verb node dev dir /Users/thomas/.node-gyp/0.12.3
gyp verb `which` succeeded for `make` /usr/bin/make
gyp info spawn make
gyp info spawn args [ 'V=1', 'BUILDTYPE=Release', '-C', 'build' ]
  c++ '-D_DARWIN_USE_64_BIT_INODE=1' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-DBUILDING_NODE_EXTENSION' -I/Users/thomas/.node-gyp/0.12.3/src -I/Users/thomas/.node-gyp/0.12.3/deps/uv/include -I/Users/thomas/.node-gyp/0.12.3/deps/v8/include -I../node_modules/nan  -Os -gdwarf-2 -mmacosx-version-min=10.5 -arch x86_64 -Wall -Wendif-labels -W -Wno-unused-parameter -fno-rtti -fno-exceptions -fno-threadsafe-statics -fno-strict-aliasing -MMD -MF ./Release/.deps/Release/obj.target/serialport/src/serialport.o.d.raw  -c -o Release/obj.target/serialport/src/serialport.o ../src/serialport.cpp
  c++ '-D_DARWIN_USE_64_BIT_INODE=1' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-DBUILDING_NODE_EXTENSION' -I/Users/thomas/.node-gyp/0.12.3/src -I/Users/thomas/.node-gyp/0.12.3/deps/uv/include -I/Users/thomas/.node-gyp/0.12.3/deps/v8/include -I../node_modules/nan  -Os -gdwarf-2 -mmacosx-version-min=10.5 -arch x86_64 -Wall -Wendif-labels -W -Wno-unused-parameter -fno-rtti -fno-exceptions -fno-threadsafe-statics -fno-strict-aliasing -MMD -MF ./Release/.deps/Release/obj.target/serialport/src/serialport_unix.o.d.raw  -c -o Release/obj.target/serialport/src/serialport_unix.o ../src/serialport_unix.cpp
../src/serialport_unix.cpp:180:7: error: redefinition of 'fd'
  int fd = open(data->path, flags);
      ^
../src/serialport_unix.cpp:156:15: note: previous definition is here
int setup(int fd, OpenBaton *data) {
              ^
../src/serialport_unix.cpp:191:5: error: non-void function 'setup' should return a value [-Wreturn-type]
    return;
    ^
../src/serialport_unix.cpp:703:20: warning: comparison of array 'device.locationId' not equal to a null pointer is always true [-Wtautological-pointer-compare]
        if (device.locationId != NULL) {
            ~~~~~~~^~~~~~~~~~    ~~~~
../src/serialport_unix.cpp:706:20: warning: comparison of array 'device.vendorId' not equal to a null pointer is always true [-Wtautological-pointer-compare]
        if (device.vendorId != NULL) {
            ~~~~~~~^~~~~~~~    ~~~~
../src/serialport_unix.cpp:709:20: warning: comparison of array 'device.productId' not equal to a null pointer is always true [-Wtautological-pointer-compare]
        if (device.productId != NULL) {
            ~~~~~~~^~~~~~~~~    ~~~~
../src/serialport_unix.cpp:712:20: warning: comparison of array 'device.manufacturer' not equal to a null pointer is always true
      [-Wtautological-pointer-compare]
[... 8 lines cut ...]
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Darwin 14.3.0
[... cut ...]
node-pre-gyp ERR! cwd /Users/thomas/work-perso/node-serialport
node-pre-gyp ERR! node -v v0.12.3
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! not ok
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/Users/thomas/work-perso/node-serialport/build/serialport/v1.7.0/Release/node-v14-darwin-x64/serialport.node --module_name=serialport --module_path=/Users/thomas/work-perso/node-serialport/build/serialport/v1.7.0/Release/node-v14-darwin-x64' (1)
```

jacobrosenthal, 2015-05-30:
On Sat, May 30, 2015 at 9:17 AM, Thomas Sarlandie <notifications@github.com>
wrote:

> gyp ERR! command "/usr/local/bin/node"
> "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js"
> "build" "--fallback-to-build" [...]

Upgrade to 1.7.1

kennethlimcp, 2015-05-30:
Isn't the latest version only 1.7.0?

sarfata, 2015-05-30:
I had a fresh checkout of the project. It is the latest.

I fixed two minor c problems in the source and it built fine after that.
Will be submitting a PR shortly.

jacobrosenthal, 2015-05-30:
https://github.com/voodootikigod/node-serialport/releases

From your paste:
"gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/
npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build"
"--module=/Users/thomas/work-perso/node-serialport/build/serialport*/v1.7.0*
/Release/node-v14-darwin-x64/serialport.node" "--module_name=serialport"
"--module_path=/Users/thomas/work-perso/node-serialport/
build/serialport/v1.7.0/Release/node-v14-darwin-x64""

sarfata, 2015-05-31:
The version I have is because `package.json` [still says 1.7.0](https://github.com/voodootikigod/node-serialport/blob/master/package.json#L3).

PR #532 (2015-05-31): Fix two small C errors that crashed build on mac os 10.10
This fixes #496 (at least on my machine).
closed 2015-05-31 by voodootikigod