serialport/node-serialport #497: XBee connection issues

bluejamesbond, 2015-04-08:
I am using this library to connect and transfer data between two xbees. Writes to the serialport works such that another xbee receives the correct data. As soon the other xbee sends data back to the current xbee, then all the future send fails from the serialport library.Your help will be much appreciated.

reconbot, 2015-04-08:
Can you post since sample code that has the issue?

bluejamesbond, 2015-04-08:
I have two xbees: one is set to coordinator API 2 and the other is router API 2. The nodejs application is connected to the coordinator. The router is connected to a pic and controlled with xctu next gen application. 

```js
var util = require('util');
var SerialPort = require('serialport').SerialPort;
var xbee_api = require('xbee-api');

var C = xbee_api.constants;

var xbeeAPI = new xbee_api.XBeeAPI({
    api_mode: 1
});

var serialport = new SerialPort("/dev/ttyAMA0", {
    baudrate: 9600,
    parser: xbeeAPI.rawParser()
});

serialport.on("open", function () {
    var frame_obj = {
        type: 0x10, 
        id: 0x01, 
        destination64: "0013A200407A25A7",
        broadcastRadius: 0x00,
        options: 0x00, 
        data: "Hello world" 
    };
    
    serialport.write(xbeeAPI.buildFrame(frame_obj));
    console.log('Sent to serial port.');

   setTimeout(function() {
         // this doesn't work if the receiving xbee sends back something 
          serialport.write(xbeeAPI.buildFrame(frame_obj));
  },  5000):
});

serialport.on('data', function (data) {
    console.log('data received: ' + data);
});

// All frames parsed by the XBee will be emitted here
xbeeAPI.on("frame_object", function (frame) {
    console.log(">>", frame);
});

// note: there are no errors coming from ' error' event
```

bluejamesbond, 2015-04-08:
I just wanted to add that the xbee does receive the  message the router xbee sends back. To help debug the error,  I placed numerous print statements in 'serialport.js` and I noticed that when I send the data the first time, the following function is called however the subsequent times, it fails to reach this line: https://github.com/voodootikigod/node-serialport/blob/master/serialport.js#L231

reconbot, 2016-04-03:
The poller for linux and osx is a c library, you'll find them user `src/` in the project. I'm closing this due to age, I'd love to hear what happened and find out if we can help you further. Sorry to take a year to get back to you!

closed 2016-04-03 by reconbot