serialport/node-serialport #501: Support .update options

jacobrosenthal, 2015-04-10:
Chrome supports an options update, letting you change baud for instance while keeping a port open
https://developer.chrome.com/apps/serial#method-update

And serial packages like Python serial do as well
http://pyserial.sourceforge.net/pyserial_api.html#serial.Serial.applySettingsDict

You cant necessarily get the same thing with an open/close because most arduino style boards wire the DTR pin so they get a reset event during the open/close.

Our immediate use case has been that Chrome DOESNT support break (yet, please help!)
https://code.google.com/p/chromium/issues/detail?id=461169

We fake a break by opening at super low baud (200) then sending a 0 byte which to a device at say 9600 is a sustained TX event that looks like a break, and then updating our baud back to 9600.

Right now we can't run the same code against node-serialport as we do on browser-serialport and we try to maintain parity so wed like to back port the update feature. 

I have a branch we're using for this already but its pretty ugly so wanted to post for discussion while I clean it up for PR.

reconbot, 2016-03-30:
We don't have docs for `update` or `set`. Did this ever land windows support?

reconbot, 2016-04-04:
This hasn't landed, we can error in the meantime or silently do nothing. I vote error.

reconbot, 2016-07-20:
The integration tests don't currently run on windows and we don't ensure we can still write to the port after a baudrate change. Once those two items have been tested I'm comfortable closing this.

codegastudio, 2016-09-27:
+1 update baudrate with documented method on mac return error " Cannot open"

serialport@4.0.1

reconbot, 2017-07-21:
This is long since solved, we currently only support updating baudRate but it's cross platform.

closed 2017-07-21 by reconbot