serialport/node-serialport #507: open () returning while fd is null

lcrocker, 2015-04-13:
Using 1.6.3 on Linux (Ubuntu 14.04). Doing the usual:

    p = serialport.SerialPort('/dev/ttyUSB0', { ... });
    p.on('open', function () {
        some_callback();
    });

In my callback, the port object is all there, but the "fd" property of the port object is still null, and attempting a write() will throw an exception. Simply waiting until fd is not null fixes the issue, but really, shouldn't open() be doing that, and not returning until the port is actually open and ready for business?

lcrocker, 2015-04-30:
Found this on several versions, and it seems to be related to running under a shell. When the program is run by manually typing "nodejs ./myprog.js ...", then all is well, or if it is run from .bashrc. It's when I try to run the program from a Gnome autostart script that it fails. "fd" comes out 0, all writes fail (and this condition persists, so I can't just wait it out).

reconbot, 2016-03-30:
@lcrocker the port should be open and writes shouldn't fail after an `open` event. We didn't support systems where the fd might be 0, but now we do!  Our latest beta `2.0.7-beta2` lands support you can read more #688

Please try it out and let us know if this fixes your issue!

closed 2016-03-30 by reconbot