serialport/node-serialport #509: Issue With Reading/Writing Between Two Beaglebones

csemiles, 2015-04-18:
I am using node-serialport to communicate between two Beaglebone Blacks connected via a Crossover RS232 cable and some serial capes.

     var serialPort;
     var connectToController = function() {
    
    serialPort = new SerialPort("/dev/ttyO0", {
        baudrate: 9600,
        parser: serialport.parsers.readline("\n")
    }, false);

    serialPort.open(function (error) {
        if(error) { 
            console.log("Error occurred: " + error + "\n");
            attemptReconnect();
        }
        else{ 
            serialPort.on('data', function(data) {
            console.log("data:" + data);
            });
        }   
    });
    serialPort.on("close", function () {
        attemptReconnect();
    });
    };

On both Beaglebones I am running code similar to above, but each of them have write capabilities as well. For some reason, I can only receive messages on one of the Beaglebones whenever I am attempting communication. Does node-serialport support read/write capability at the same time, or is there something wrong with my code?

jacobrosenthal, 2015-04-18:
Where is the code that actually sends something. You're using a new line parser, are you sure that the .write of data ends with a newline? onData wont return otherwise. Do you have a common ground?

csemiles, 2015-04-18:
Thank you for the quick reply.
In Beaglebone B1, 

    var serialport = require("serialport");
    var SerialPort = serialport.SerialPort;
    var serialPort;

    var connectToController = function() {
    
    
    serialPort = new SerialPort("/dev/ttyO0", {
        baudrate: 9600,
        parser: serialport.parsers.readline("\n")
    }, false);

    
    //attempt to open the port
    serialPort.open(function (error) {
        
        if(error) { 
            console.log("Error occurred: " + error + "\n");
            attemptReconnect();
        }
        else{ 
            
            
            console.log("Success! Connected to the controller.\n");
            
           
            serialPort.on('data', function(data) {
                
                console.log("data:" + data);
                serialPort.write("HELLO FROM B1\n");
        
            });
        }
        
    });
    
    serialPort.on("close", function () {
        console.log("The serial channel to the controller has closed.\n");
        attemptReconnect();
    });
    
    };

    var attemptReconnect = function () {
     setTimeout(function(){
       console.log("Reconnecting to controller");
       connectToController();
      }, 5000);
    };

    connectToController();

Beaglebone B2 is very similar, except that it sends the first data message

    var serialport = require("serialport");
    var SerialPort = serialport.SerialPort;
    var serialPort;

    var connectToController = function() {
    
    
    serialPort = new SerialPort("/dev/ttyO0", {
        baudrate: 9600,
        parser: serialport.parsers.readline("\n")
    }, false);

    
    //attempt to open the port
    serialPort.open(function (error) {
        
        if(error) { 
            console.log("Error occurred: " + error + "\n");
            attemptReconnect();
        }
        else{ 
            console.log("Success! Connected to the controller.\n");
            serialPort.write("HELLO FROM B2\n");
           
            serialPort.on('data', function(data) {
                
                console.log("data:" + data);
                serialPort.write("HELLO FROM B2\n");
        
            });
        }
        
    });
    
    serialPort.on("close", function () {
        console.log("The serial channel to the controller has closed.\n");
        attemptReconnect();
    });
    
    };

    var attemptReconnect = function () {
     setTimeout(function(){
       console.log("Reconnecting to controller");
       connectToController();
      }, 5000);
    };

    connectToController();

I expect B2 to send the first message, which is received by B1, and then B1 sends the second message, and so the two should keep communicating. What actually happens is B1's first message is never received by B2. As for the common ground question, I am not sure? At the moment I have the Beaglebones plugged into two computers via USB.

closed 2016-04-06 by reconbot