serialport/node-serialport #512: serialport + nw.js v0.12.1 (OSX)

julianh2o, 2015-04-21:
I've read through most of this issue but haven't had much luck getting any of the solutions working.. https://github.com/voodootikigod/node-serialport/issues/374

Running OSX 10.10.3
I've made various attempts at both nw.js 0.12.1 and 0.8.6
Python 2.7.6
node v0.12.2
npm  v2.7.5

The error is as follows.. (0.8.6)
Uncaught Error: Cannot find module '[...]/MyProject/node_modules/serialport/build/serialport/v1.6.3/Release/node-webkit-v11-darwin-ia32/serialport.node' 

or .. (0.12.1)

Uncaught Error: Cannot find module '[...]/MyProject/node_modules/serialport/build/serialport/v1.6.3/Release/node-webkit-v43-darwin-x64/serialport.node' 

**Trying to compile with node-pre-gyp has failed in various permutations: (one such..)**
    serialport $ ./node_modules/node-pre-gyp/bin/node-pre-gyp build --runtime-node-webkit --target=0.8.6
    node-pre-gyp info it worked if it ends with ok
    node-pre-gyp info using node-pre-gyp@0.5.19
    node-pre-gyp info using node@0.12.2 | darwin | x64
    child_process: customFds option is deprecated, use stdio instead.
    node-pre-gyp ERR! build error
    node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp configure build --name=serialport --configuration=Release --module_name=serialport --version=1.6.3 --major=1 --minor=6 --patch=3 --runtime=node --node_abi=v8-3.11 --target=0.8.6 --platform=darwin --target_platform=darwin --arch=x64 --target_arch=x64 --module_main=./serialport --host=https://node-serialport.s3.amazonaws.com/ --module_path=/Users/julian/Documents/Programming/LEDControl/node_modules/serialport/build/serialport/v1.6.3/Release/v8-3.11-darwin-x64 --remote_path=./serialport/v1.6.3/Release/ --package_name=v8-3.11-darwin-x64.tar.gz --staged_tarball=build/stage/serialport/v1.6.3/Release/v8-3.11-darwin-x64.tar.gz --hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.6.3/Release/ --hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.6.3/Release/v8-3.11-darwin-x64.tar.gz' (Error: spawn node-gyp ENOENT)
    node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/Users/julian/Documents/Programming/LEDControl/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:70:29)
    node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:107:17)
    node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1072:12)
    node-pre-gyp ERR! stack     at child_process.js:1144:20
    node-pre-gyp ERR! stack     at process._tickCallback (node.js:355:11)
    node-pre-gyp ERR! stack     at Function.Module.runMain (module.js:503:11)
    node-pre-gyp ERR! stack     at startup (node.js:129:16)
    node-pre-gyp ERR! stack     at node.js:814:3
    node-pre-gyp ERR! System Darwin 14.3.0
    node-pre-gyp ERR! command "node" "/Users/julian/Documents/Programming/LEDControl/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp" "build" "--runtime-node-webkit" "--target=0.8.6"
    node-pre-gyp ERR! cwd /Users/julian/Documents/Programming/LEDControl/node_modules/serialport
    node-pre-gyp ERR! node -v v0.12.2
    node-pre-gyp ERR! node-pre-gyp -v v0.5.19
    node-pre-gyp ERR! not ok
    Failed to execute 'node-gyp configure build --name=serialport --configuration=Release --module_name=serialport --version=1.6.3 --major=1 --minor=6 --patch=3 --runtime=node --node_abi=v8-3.11 --target=0.8.6 --platform=darwin --target_platform=darwin --arch=x64 --target_arch=x64 --module_main=./serialport --host=https://node-serialport.s3.amazonaws.com/ --module_path=/Users/julian/Documents/Programming/LEDControl/node_modules/serialport/build/serialport/v1.6.3/Release/v8-3.11-darwin-x64 --remote_path=./serialport/v1.6.3/Release/ --package_name=v8-3.11-darwin-x64.tar.gz --staged_tarball=build/stage/serialport/v1.6.3/Release/v8-3.11-darwin-x64.tar.gz --hosted_path=https://node-serialport.s3.amazonaws.com/serialport/v1.6.3/Release/ --hosted_tarball=https://node-serialport.s3.amazonaws.com/serialport/v1.6.3/Release/v8-3.11-darwin-x64.tar.gz' (Error: spawn node-gyp ENOENT)

**Here's another..**
    serialport $ ./node_modules/node-pre-gyp/bin/node-pre-gyp build --runtime-node-webkit --target=0.12.1
    node-pre-gyp info it worked if it ends with ok
    node-pre-gyp info using node-pre-gyp@0.5.19
    node-pre-gyp info using node@0.12.2 | darwin | x64
    node-pre-gyp ERR! UNCAUGHT EXCEPTION
    node-pre-gyp ERR! stack Error: Unsupported target version: 0.12.1
    node-pre-gyp ERR! stack     at get_node_abi (/Users/julian/Documents/Programming/LEDControl/node_modules/serialport/node_modules/node-pre-gyp/lib/util/versioning.js:27:19)
    node-pre-gyp ERR! stack     at Object.module.exports.evaluate (/Users/julian/Documents/Programming/LEDControl/node_modules/serialport/node_modules/node-pre-gyp/lib/util/versioning.js:133:21)
    node-pre-gyp ERR! stack     at build (/Users/julian/Documents/Programming/LEDControl/node_modules/serialport/node_modules/node-pre-gyp/lib/build.js:22:27)
    node-pre-gyp ERR! stack     at Object.self.commands.(anonymous function) [as build] (/Users/julian/Documents/Programming/LEDControl/node_modules/serialport/node_modules/node-pre-gyp/lib/node-pre-gyp.js:48:37)
    node-pre-gyp ERR! stack     at run (/Users/julian/Documents/Programming/LEDControl/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp:77:30)
    node-pre-gyp ERR! stack     at Object.<anonymous> (/Users/julian/Documents/Programming/LEDControl/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp:130:1)
    node-pre-gyp ERR! stack     at Module._compile (module.js:460:26)
    node-pre-gyp ERR! stack     at Object.Module._extensions..js (module.js:478:10)
    node-pre-gyp ERR! stack     at Module.load (module.js:355:32)
    node-pre-gyp ERR! stack     at Function.Module._load (module.js:310:12)
    node-pre-gyp ERR! System Darwin 14.3.0
    node-pre-gyp ERR! command "node" "/Users/julian/Documents/Programming/LEDControl/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp" "build" "--runtime-node-webkit" "--target=0.12.1"
    node-pre-gyp ERR! cwd /Users/julian/Documents/Programming/LEDControl/node_modules/serialport
    node-pre-gyp ERR! node -v v0.12.2
    node-pre-gyp ERR! node-pre-gyp -v v0.5.19
    node-pre-gyp ERR! This is a bug in `node-pre-gyp`.
    node-pre-gyp ERR! Try to update node-pre-gyp and file an issue if it does not help:
    node-pre-gyp ERR!     <https://github.com/mapbox/node-pre-gyp/issues>

julianh2o, 2015-04-21:
The NPM install seemed to work alright...

$ npm install serialport
npm WARN package.json wifiledcontrol@0.0.1 No repository field.
\
> serialport@1.6.3 install /Users/julian/Documents/Programming/LEDControl/node_modules/serialport
> node-pre-gyp install --fallback-to-build

child_process: customFds option is deprecated, use stdio instead.
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/serialport_poller.o
  SOLINK_MODULE(target) Release/serialport.node
  SOLINK_MODULE(target) Release/serialport.node: Finished
  COPY /Users/julian/Documents/Programming/LEDControl/node_modules/serialport/build/serialport/v1.6.3/Release/node-v14-darwin-x64/serialport.node
  TOUCH Release/obj.target/action_after_build.stamp
serialport@1.6.3 node_modules/serialport
├── bindings@1.2.1
├── sf@0.1.7
├── nan@1.7.0
├── debug@2.1.3 (ms@0.7.0)
└── optimist@0.6.1 (wordwrap@0.0.2, minimist@0.0.10)

voodootikigod, 2015-05-09:
This is due to the way nw.js changes the Node-ABI. This is not an issue with serialport, review this for more information: https://github.com/nwjs/nw.js/wiki/Using-Node-modules#3rd-party-modules-with-cc-addons

Closing

closed 2015-05-09 by voodootikigod