serialport/node-serialport #513: Problems with installation and using on win7 x64 and nw.js 0.12.1

jet112, 2015-04-21:
I have installed the latest version of node.js, then I have downloaded and unpacked the latest version of node-webkit (nw.js) to my project folder. Then I tried to install node-serialport using npm: 
> C:\my_app > npm install serialport

Then I have tried to run a simple script: 
```<script>var SerialPort = require("serialport").SerialPort</script>```

but node-webkit has flashed an error: 
> Uncaught Error: Cannot find module
> C:\my_app\node_modules\serialport\build\serialport\v1.6.3\Release\node-webkit-v43-win32-x64\serialport.node

I have noticed that there was a wrong release folder name during installation. So, I have renamed folder ```...node_modules\serialport\build\serialport\v1.6.3\Release\node-v14-win32-x64``` to ```...node_modules\serialport\build\serialport\v1.6.3\Release\node-webkit-v43-win32-x64``` and tried to run my script again. But now node-webkit flashes an error:
> module.js:346 Uncaught Error: Module did not self-register.

Help me please! How can I fix it? And how to properly install such modules for node-webkit using npm?

jedthehumanoid, 2015-04-22:
If you just install native modules with npm, they will be build for node, not nw-js.

https://github.com/nwjs/nw.js/wiki/Using-Node-modules#3rd-party-modules-with-cc-addons

voodootikigod, 2015-05-09:
This is due to the way nw.js changes the Node-ABI. This is not an issue with serialport, review this for more information: https://github.com/nwjs/nw.js/wiki/Using-Node-modules#3rd-party-modules-with-cc-addons

Closing

closed 2015-05-09 by voodootikigod