serialport/node-serialport #516: Format of data messages

vallettea, 2015-04-28:
Hello,

thx for this awesome work. 
I'm using this library to communicate with a 3G dongle. It works fine however when using

```
serialPort.on('data', function(data) {
    
});
```

I have troubles working with `data`. When i use `data.toString()` I can see the AT commands but it seems it is an object because 

```
Object.keys(message).forEach(function(key){
		console.log(key, message[key]);
})
```

outputs this:

```
0 65
1 84
2 43
3 67
4 77
5 71
6 76
7 61
8 34
9 65
10 76
11 76
12 34
13 13
14 13
15 10
16 43
17 67
18 77
19 71
20 76
21 58
22 32
23 48
24 44
25 34
26 82
27 69
28 67
29 32
30 82
31 69
32 65
33 68
34 34
35 44
36 34
37 43
38 51
39 51
40 54
41 55
42 49
43 51
44 53
45 56
46 57
47 52
48 51
49 34
50 44
51 44
52 34
53 49
54 53
55 47
56 48
57 52
58 47
59 50
60 56
61 44
62 49
63 49
64 58
65 48
66 55
67 58
68 50
69 52
70 43
71 48
72 56
73 34
74 13
75 10
76 84
77 117
78 32
79 118
80 97
81 115
82 32
83 118
84 111
85 105
86 114
87 13
88 10
89 13
90 10
91 79
92 75
93 13
94 10
length 95
parent <SlowBuffer 41 54 2b 43 4d 47 4c 3d 22 41 4c 4c 22 0d 0d 0a 2b 43 4d 47 4c 3a 20 30 2c 22 52 45 43 20 52 45 41 44 22 2c 22 2b 33 33 36 37 31 33 35 38 39 34 33 22 2c ...>
offset 0
```

I know there is all the structure of pdu sms message of this kind `06918919015000240C9189194238148900003110211052254117CAB03D3C1FCBD3703AA81D5E97E7A079D93D2FBB00` but I don't know how to parse it.

thanks in advance

voodootikigod, 2015-04-28:
The .on('data', function (message)) { }) callback returns a buffer object
https://nodejs.org/api/buffer.html i think you are ok if you check through
the documentation -- let me know if you need anything further.

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

vallettea, 2015-04-28:
ok thx

closed 2015-04-28 by vallettea