serialport/node-serialport #517: Cannot set custom baudrate on Linux.

drscaon, 2015-04-29:
I tried to apply this following code into the node-serialport module, but still with no success...
https://jim.sh/ftx/files/linux-custom-baudrate.c

I saw you left a message on the code of serialport_unix.cpp, however it is still not clear to me:
**// On linux you can alter the meaning of B38400 to mean a custom baudrate...**

Another option could be to change termios.h for asm/termios.h and use a termios2 struct but that would break many other things in chain.

jacobrosenthal, 2015-07-17:
So what custom baud could you not acheive? On which linux distro and architecture, node and serialport versions?

That comment appears to mean you could alter the cpp (not js) source code (and rebuild source) to change B38400 to say B1500000

Yeah large refactors of platform src are unilkely currently as were currently in a node working group to try to solve some of these cross platform hardware issues.

drscaon, 2015-07-17:
Thank you jacobronsenthal for your answer. 
I did a similar workaround, now I call the new version of termios by the re-compiled cpp.

closed 2015-07-17 by jacobrosenthal