serialport/node-serialport #523: old version of node-pre-gyp being install by npm with serialport

mMerlin, 2015-05-09:
I have encountered an issue getting node-serialport to install 'cleanly' using npm.

Environment info:
$ uname -srvpio
Linux 3.19.5-200.fc21.x86_64 #1 SMP Mon Apr 20 19:51:56 UTC 2015 x86_64 x86_64 GNU/Linux
$ node -v
v0.10.38
$ npm -v
2.7.6

The actual
```bash
$ npm install serialport
```
or
```bash
$ npm install
```
using a package.json file including a dependency for serialport (with or without a version) complete without issues, but a following
```bash
$ npm list
```
complains about invalid dependency versions.  The full list is fairly long, but
```
$ npm list string_decoder
/home/phil/development/projects/test
└─┬ serialport@1.6.3
  └─┬ node-pre-gyp@0.5.19 invalid
    └─┬ tar-pack@2.0.0
      └─┬ readable-stream@1.0.27-1 invalid
        └── string_decoder@0.10.25-1  invalid

npm ERR! invalid: node-pre-gyp@0.5.19 /home/phil/development/projects/test/node_modules/serialport/node_modules/node-pre-gyp
npm ERR! invalid: readable-stream@1.0.27-1 /home/phil/development/projects/test/node_modules/serialport/node_modules/node-pre-gyp/node_modules/tar-pack/node_modules/readable-stream
npm ERR! invalid: string_decoder@0.10.25-1 /home/phil/development/projects/test/node_modules/serialport/node_modules/node-pre-gyp/node_modules/tar-pack/node_modules/readable-stream/node_modules/string_decoder
```
Shows the invalid section.

My research shows this to be related to the package.json entry
```
"bundleDependencies": [
  "node-pre-gyp"
],
```
The repository does not actually (seem to) have the package bundled.  However, symptoms are that the published npm registry does not properly 'fall back' for this case.  I do not know where the v0.5.19 is comming from, but I have seen another case #506 here where someone got the same version.  From #494 and the link to https://github.com/npm/npm/issues/6556, this could be a problem with npm versions used a different times, for different packages.  Since node-pre-gyp is nolonger bundled with serialport, removing that entry from package.json should clean things up.  If the registry gets updated too.

I say that the problem is with the published registry, because bypassing that avoids the errors.  Using
```bash
$ npm install voodootikigod/node-serialport
```
or
```bash
$ npm install voodootikigod/node-serialport#v1.6.1
```
to specify a specific version end up with node-pre-gyp@0.6.7, and no error messages in npm list

voodootikigod, 2015-05-09:
Ummm... serial port requires node-pre-gyp of `"node-pre-gyp": "0.6.x"` so... its not us...

voodootikigod, 2015-05-09:
It has been that since 1.5.x (possibly before) https://github.com/voodootikigod/node-serialport/blob/94b7bdcc087b820e10913c483bab3e2c54d103f2/package.json#L49

mMerlin, 2015-05-09:
Agreed, that package.json has been specifying a dependency of node-pre-gyp 0.6.x for long time.  However, it has also been saying that node-pre-gyp is a bundledependency.   That second piece seems to be causing (or at least triggering) the problem (my environment) when it is not actually bundled in the repo.  If I install directly from the repo, I get a warning about be missing bundle
```
npm WARN installMany node-pre-gyp was bundled with serialport@1.6.3, but bundled package wasn't found in unpacked tree
```
and end up with the latest version.  If if install through the published registry, I do not get that warning, and end up with node-pre-gyp@0.5.19

Why is it marked as bundled?  Why is it not bundled in the repo when specified?

voodootikigod, 2015-05-09:
We need pre-gyp in order to provide pre-compiled binaries.  Better question would be why isn't the other package using the latest node-pre-gyp?

mMerlin, 2015-05-11:
What it is used for is separate from whether it is (marked as) bundled with the package.  Would the functionality work (the same) if package.json had the 0.6.x dependency, but did NOT have the bundledependecies?

The starting 'other package' was johnny-five.  It does need node-pre-gyp at all for its own installation or use.  Starting, because the symptoms apply when directly installing node-serial port in an empty project, where there is no 'other package' to consider.

mMerlin, 2015-05-11:
Looking at the recent commits, I do not see anything obviously related, but a simple
```bash
npm install serialport
```
works in my environment now.  Unless it was fixed by something in the latest yum update.
```bash
npm list nod-pre-gyp 
```
now shows that serialport@1.7.1 and node-pre-typ@0.6.7 were installed.

Without more information, I would have to say that the latest tarball is actually including the node-pre-gyp bundle dependency that package.json says it is.

Or the later npm version fixed the problem.  Comparing the latest working, and earlier failing node_modules/serialport/package.json files (auto generated by npm when using the registry?), I see that _npmVersion property has changed from 2.5.1 to 2.8.3

closed 2015-05-11 by mMerlin