serialport/node-serialport #529: Callback not called on successful writes

fsamir, 2015-05-27:
The function `write(buffer, callback)` doesn't not call the callback when it succeeds, only when there is an error.
Given the asynchronous nature of node, shouldn't it call the callback with an undefined error, when it succeeds?

Thanks,
Franklin

djMax, 2015-06-02:
It also appears to call it multiple times, which is very weird.

vird, 2015-09-18:
Try this
https://github.com/voodootikigod/node-serialport/compare/master...vird:patch-1

jacobrosenthal, 2015-10-13:
What platforms and version?  I tested and I got a single callback on successful write.
osx yosemite (iojs-v2.3.4), serialport 1.7.4 and 2.0.0
Raspberry pi (v0.10.40) serialport 1.7.4 and 2.0.

reconbot, 2016-04-06:
We have a lot of tests around this, it's not normal behavior so you might have found a strange combination of things. We also have new errors that happen in bad situations (eg, the port is closing, or not yet open). As a reminder you can only write when the port is open.

I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to reopen.

closed 2016-04-06 by reconbot