serialport/node-serialport #530: Windows 10 SerialPort.list uses cached list

matthova, 2015-05-28:
Steps:
1. Use Windows 10
2. Insert a serial device(s)
3. Remove serial device(s)
4. Restart the computer if you want

Running the following code after unplugging 2 serial devices logs as follows:
var serialPort = require("serialport");
serialPort.list(function (err, ports) {
  ports.forEach(function(port) {
  	console.log("port!\n", port, "\n\n");
  });
});

C:\Users\matthova\Desktop\serial_example>node app.js
port!
 { comName: 'COM4',
  manufacturer: 'Microsoft',
  serialNumber: '',
  pnpId: 'USB\\VID_2341&PID_0010\\5&330350CB&0&1',
  locationId: '',
  vendorId: '',
  productId: '' }

port!
 { comName: 'COM3',
  manufacturer: 'Microsoft',
  serialNumber: '',
  pnpId: 'USB\\VID_2341&PID_0010\\5&330350CB&0&5',
  locationId: '',
  vendorId: '',
  productId: '' }

The serial device list retains devices that have been unplugged. This bug has been repeated on multiple Windows 10 machines and on node versions 0.10.35 and 0.12.4. For reference, the version of Windows 10 I am using is en_windows_10_enterprise_technical_preview_9926_x64_dvd_6244735

JayBeavers, 2015-07-17:
Is this specific to Windows 10?  I've seen similar behavior on bluetooth serial devices on Windows 8.1 as well.

This doesn't sound like it's a bug with node-serialport as much as the OS itself.

matthova, 2015-08-17:
@JayBeavers Is this a repeating incident or a one off bug? With Windows 10 it's an issue every time.

reconbot, 2016-03-27:
@JayBeavers @matthova any update on this, I don't have a win10 box to test with. I can say we're using the standard apis for polling serialports, and we're not caching them in `node-serialport` itself.

reconbot, 2016-04-08:
I can't reproduce this behavior on windows 7 or 10. I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to reopen.

closed 2016-04-08 by reconbot