serialport/node-serialport #533: Open only 4 ports truly

psy21d, 2015-05-31:
this is not full listing, some tests.

var SerialPort = require("serialport");
		var self = this;
		var _serialPorts = []
		var num = 0;

		SerialPort.list(function (err, ports) {
		  ports.forEach(function(port) {
		    console.log(port.comName);
		    console.log(port.pnpId);
		    console.log(port.manufacturer);
		    	setTimeout(function () {
		    			serialPort = new SerialPort.SerialPort(port.comName, {
		                        baudrate: 9600,
		                        bufferSize : 2048
	                      },true, this._error );
		    			serialPort.on('data', function(data) {
						    console.log('data received: ' + data);
						});
		    			serialPort.write("ATI\n", function(err, results) {console.log("write ok")});
		    			_serialPorts.push(serialPort);
					console.log("port %s go open",port.comName)
				},num*400);
				num++;
		  });
		});

javascript, iojs -v 
v1.8.1

Can not read from ports opened (try to opened > 4 items), can open only 4 ports.
next ports only try to open and not work at all. why?

When I close ports, this can work, but only for first 4 ports!!
Fifth comport only open and not read at all, and sixth++ ports return error "port not open" when try to close it!
If I open another script instance, remaining ports can be opened and only four too.

It is very big problem to use this.

psy21d, 2015-05-31:
Another big problem: fourth port not actually closed when i command to close it:
If start another script, then that fuckin:

port COM26 go open
port undefined open error Error: Opening \\.\COM26: Unknown error code 170
closing error Error: Serialport not open. port COM26

port COM29 go open
port undefined open error Error: Opening \\.\COM29: Unknown error code 170
closing error Error: Serialport not open. port COM29

port COM37 go open
port undefined open error Error: Opening \\.\COM37: Unknown error code 170
closing error Error: Serialport not open. port COM37

port COM28 go open
port undefined open error Error: Opening \\.\COM28: Access denied
closing error Error: Serialport not open. port COM28

COM29 not busy and cannot accessed. why?
First script can not work with fourth port and just fuckup it.

All devices good, all devices answering normally, problem in node-serialport and I not understood how to fix it recently.

corelgott, 2015-06-17:
I am hitting this problem too! I need to communicate with 4 serial ports simultaneously. When using 3 everything works just fine. But when opening any 4th port - all communication on all ports stop for good.

Where does this 4-Port limit come from?
(WTH!?)

corelgott, 2015-06-20:
Hi,

i've tracked down the problem a little more… in `src/serialport.cpp` in line 138 the EIO_Open routine of `SerialPortWin.cpp` is pushed onto the libuv execution queue.

```uv_queue_work(uv_default_loop(), req, EIO_Open, (uv_after_work_cb)EIO_AfterOpen);```

For some reason the 4th item in the queue waits until the other are finished. I'd guess this is either a problem within libuv itself or with the way of usage. It might help to use a different event loop or, if possible, to change some lib-uv resource setting to spawn more threads. But since I don't know any of the libuv api and haven't worked with it up to now, this is only a guess.

psy21d, 2015-06-21:
Well, I was not able to get into Cpp code, but:

I fully understand what the problem is.
It is there and it is in that port waiting for the rest of the fourth.
But he will never be found, for the simple reason that all the ports needed to me! 10, 20, 30 ports.
When the previous ports closed, of course it is got taken, but with closing ports there also has problem.

s'ry 4 m bd g0gl E.

psy21d, 2015-06-21:
Only one truly way is to spawn a thread for each port.

corelgott, 2015-06-21:
Hi psy21d,

if I read the cpp code correctly it hands every command you wants to perform on the serial port a  threadpool. So if you call open on the js side, the cpp with hand of the open-procedure to the `uv_queue_work` of lib-uv. Which will use / create a thread to perform the requested action – as described here: http://docs.libuv.org/en/latest/threadpool.html – The default size of the thread pool used is 4 (just found that in the docs) I'll see if I can mock together a test scenario where I increase the `UV_THREADPOOL_SIZE `. If I have any news on this, I'll provide feedback!

corelgott, 2015-06-22:
Hi there,

I think I found a workaround / the fix for the described problem. It is in deed the `UV_THREADPOOL_SIZE` which causes this behavior.

To work around this, I had to set the environment variable via console
```
shell> SET UV_THREADPOOL_SIZE = 16
```
which then allows me to communicate on **4 ports simultaneously**! I've tested this on Win7x64.
Setting it via `process.env.UV_THREADPOOL_SIZE = 16` **did not work** for me!

Even though this isn't a bug in node-serialport it self, I think it is worth mentioning this behavior within the documentation.

psy21d, 2015-06-22:
Good.  SET UV_THREADPOOL_SIZE = 16 via cmd.exe does not make any effect.
Win7x64
Need to reboot? How to test result fast?

corelgott, 2015-06-22:
You need to set this value in the command line you execute your node-app with. Settings this value is only true for the process you set it in or it's sub-processes. If you are using an IDE you need to set it within itself. You can set the variable for you whole system as described here: http://tinyurl.com/3n4zy

I've managed to get it to work like this on Win7x64, Win7x32 & within node_webkit

psy21d, 2015-06-22:
Yes, I set this variable in system path, of course

jacobrosenthal, 2015-07-17:
Thanks for tracking down guys. If you can think of a way this should be documented, PR welcome.

closed 2015-07-17 by jacobrosenthal

camposcristianeze, 2015-10-10:
Hello guys,

I'm experiencing something similar but only by opening and closing 4 times the same COM port. I've tracked the code reaching to 
``` uv_queue_work(uv_default_loop(), req, EIO_Close, (uv_after_work_cb)EIO_AfterClose); ```
I'm testing an app where the serialport could be disconected, so I want to reconnect it always.  
I don't want to open more threads, just figure out why at the fourth time I'm reaching EIO_Close, it's not executing.
I appreciate any suggestion!

Thanks

Thanks!

psy21d, 2015-10-11:
**camposcristianeze**, it was not possible to solve this problem. For this reason, I completely abandoned this node-serialport.
How to work with him if he does not allows you to do at least anything?

psy21d, 2015-10-11:
The only thing that can be done here - to open each port in a separate thread node-js, then erase when you want to take com port.
node-serialport currently allows only one port and only once. All these "system settings" does not help you.