serialport/node-serialport #535: Fails to install on Raspberry 2

maxkattner, 2015-05-31:
Sadly it seems to be impossible, to install serialport on (my) Raspberry Pi 2 running raspbian 7.
I tried several variations of using sudo and/or different versions. (which where mentioned in other issues)

This is the output I get when trying `npm install serialport`
```
> serialport@1.7.3 install /home/pi/node_modules/serialport
> node-pre-gyp install --fallback-to-build

child_process: customFds option is deprecated, use stdio instead.
child_process: customFds option is deprecated, use stdio instead.
make: Entering directory '/home/pi/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
../src/serialport_unix.cpp: In function 'int setup(int, OpenBaton*)':
../src/serialport_unix.cpp:186:7: error: declaration of 'int fd' shadows a parameter
serialport.target.mk:85: recipe for target 'Release/obj.target/serialport/src/serialport_unix.o' failed
make: *** [Release/obj.target/serialport/src/serialport_unix.o] Error 1
make: Leaving directory '/home/pi/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1067:12)
gyp ERR! System Linux 3.18.11-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/node_modules/serialport/build/serialport/v1.7.3/Release/node-v14-linux-arm/serialport.node" "--module_name=serialport" "--module_path=/home/pi/node_modules/serialport/build/serialport/v1.7.3/Release/node-v14-linux-arm"
gyp ERR! cwd /home/pi/node_modules/serialport
gyp ERR! node -v v0.12.1
gyp ERR! node-gyp -v v1.0.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/node_modules/serialport/build/serialport/v1.7.3/Release/node-v14-linux-arm/serialport.node --module_name=serialport --module_path=/home/pi/node_modules/serialport/build/serialport/v1.7.3/Release/node-v14-linux-arm' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:1008:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1080:5)
node-pre-gyp ERR! System Linux 3.18.11-v7+
node-pre-gyp ERR! command "node" "/home/pi/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/node_modules/serialport
node-pre-gyp ERR! node -v v0.12.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! not ok 
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/node_modules/serialport/build/serialport/v1.7.3/Release/node-v14-linux-arm/serialport.node --module_name=serialport --module_path=/home/pi/node_modules/serialport/build/serialport/v1.7.3/Release/node-v14-linux-arm' (1)
npm ERR! Linux 3.18.11-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "serialport"
npm ERR! node v0.12.1
npm ERR! npm  v2.5.1
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.3 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
```

Please don't hesitate to tell me, if I'm doing something completely wrong, but this error seems to be very odd.

maxkattner, 2015-06-01:
Okay, got it to work now with version 1.6.1

jywarren, 2015-06-04:
I'm having trouble with this as well, on v1.7.4, as shown in output here: https://gist.github.com/jywarren/c877dc744c7e342bd025

Do you know what about 1.6.1 makes it work? Is it something we could fix? I don't directly call serialport, but use it via the firmata node library. So I can't cleanly specify 1.6.1.

jywarren, 2015-06-04:
I'm using node v0.10.36, too, so it doesn't seem to be the issue elsewhere in this tracker which was fixed by updating to v0.10.16.

jywarren, 2015-06-04:
It looks like this is the key error:

> make: *** No rule to make target '../.node-gyp/0.10.36/common.gypi', needed by 'Makefile'.  Stop.

So, I'll check in on the node-pre-gyp project.

jywarren, 2015-06-04:
It seems like it's more about the permissions where I'm running `npm install serialport`. 

````
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/0.10.36"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/local/lib/node_modules/serialport/.node-gyp"
make: Entering directory '/usr/local/lib/node_modules/serialport/build'
make: *** No rule to make target '../.node-gyp/0.10.36/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/usr/local/lib/node_modules/serialport/build'
````

gbox3d, 2015-07-10:
it to work now with version 1.6.2 too

jaumard, 2015-08-03:
Not working for me too ! 
1.6.2 and 1.6.3 are working but nothing above

niko34, 2015-08-15:
did you try --unsafe-perm option ?
sudo npm install --unsafe-perm serialport

jaumard, 2015-08-15:
@niko34 it's working with --unsafe-perm but what this option does exactly ?

niko34, 2015-08-15:
I cannot tell you exactly what happens, but when i had the same error, i noticed this message : "gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/0.12.0". Even if i used "sudo npm...."
I had the idea it was related to a right problem. I found this thread : http://stackoverflow.com/questions/18136746/npm-install-failed-with-cannot-run-in-wd

It tells "Set the unsafe-perm flag to run scripts with root privileges.". May-be using sudo isn't enough to give scripts root privileges....

We need an expert to tell you more...

jacobrosenthal, 2015-09-12:
I just installed 1.7.4  (node v0.10.40) on a pi without issue or any extraneous flags.

jacobrosenthal, 2015-09-12:
Is there a reason our docs link to a specific node version. What if we do the nodesource setup script
```
curl -sL https://deb.nodesource.com/setup | sudo bash 
```

closed 2015-09-14 by jacobrosenthal