serialport/node-serialport #536: output is being truncated when data is received from the serial device

c0debreaker, 2015-06-01:
I used minicom and screen utilities to connect to my serial device. I can see the messages being sent by my serial device in single line.

However, when I use SerialPort, the strings are being displayed in multiple lines. Sometimes one line but most of the time it's multiple lines. The data being sent by the serial deviceis really short, maybe about 16 characters only. Here is my code

```
serialPort.on("open", function () {
  console.log('open');
  serialPort.on('data', function(data) {
    console.log('data received: ' + data);
  });
});
```

Here is a screenshot while I was running my nodejs application( http://i.imgur.com/Lul3s23.png ). You will see, it's being displayed multiple lines. If you look at the topmost, using minicom or screen, I will only see 16XK5905221310515110006D in one line.

Is there something I need to change on my code?

c0debreaker, 2015-06-01:
It's ok now, I figured out a way to combine them. I used chars 13 & 10 as the terminator.

closed 2015-06-01 by c0debreaker

Moussawi7, 2015-07-04:
Hi @c0debreaker ,
In fact i encountered exactly the same issue,
where you terminate the string with the characters(13,10) ?.
Thank you.

c0debreaker, 2015-07-04:
This is how I'm doing it

```
    for (i = 0; i < data.length; i++) {
        if (data[i] != 10 || data[i] != 13) {
          buffer += String.fromCharCode(data[i]);
        }
        if (data[i] == 13) {
            if (buffer.length > 0) {
                //console.log('buffer: '+ buffer.replace(/\n|\r/g, ''));
                reading = buffer.replace(/\n|\r/g, '').toString();
                // All the magic here
		//console.log('reading', reading);
```