serialport/node-serialport #538: Unable to require serialport in Electron's main.js file

a-giovani17, 2015-06-04:
I have an Electron desktop app that requires some nodejs packages including serialport. But only serialport is failed to require.

I included it like this:

var SerialPort = require('serialport').SerialPort;
var sp = require('serialport');

But when I run it on Electron, it shows following error:

Error: Cannot find module 'C:\xampp\htdocs\simplemail-desktop\node_modules\serialport\build\serialport\v1.7.4\Release\node-v43-win32-ia32\serialport.node'

I've check that the folder actually doesn't exists. When I installed serialport, it created folder 'node-v14-win32-ia32' instead of 'node-v43-win32-ia32'. Maybe there is something wrong in the installation process. What should I do?

voodootikigod, 2015-06-04:
I just got my electron today will investigate!

what version of node are you using?

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

bongia, 2015-06-04:
* Node version: 0.12.4
* Electron version: 0.27.2

The code is simply a boilerplate of electron:

```javascript
var app = require("app");
var BrowserWindow = require("browser-window");
var mainWindow = null;

app.on("window-all-closed", function () {
  if (process.platform !== "darwin") {
    app.quit();
  }
});

app.on("ready", function () {
  mainWindow = new BrowserWindow({
    width: 960,
    height: 600
  });
  mainWindow.loadUrl("file://" + __dirname + "/index.html");
  mainWindow.on("closed", function () {
    mainWindow = null;
  });

  mainWindow.openDevTools();

  var SerialPort = require("serialport").SerialPort;
  var serialPort = new SerialPort("TODO-TODO-EL-MEXICO!", {
    baudrate: 57600
  });

});
```

landuomu, 2015-06-09:
Same here

I am using Ubuntu 14.04 LTS. I got this when I require serialport when i run in electron.

[3512:0609/164117:INFO:CONSOLE(326)] "Uncaught Error: Cannot find module './node_modules/serialport/build/serialport/v1.4.10/Release/node-v43-linux-x64/serialport.node'", source: module.js (326)

I didn't found this folder but I found './node_modules/serialport/build/serialport/v1.4.10/Release/node-v11-linux-x64/serialport.node'

Node version: 0.10.25
Electron version: 0.27.2

a-giovani17, 2015-06-09:
Actually when I renamed the folder to ''node-v43-win32-ia32', this error shown:

Uncaught Exception:
Error: Invalid Access to Memory Location

maybe there is something wrong in the native module (serialport.node)

landuomu, 2015-06-09:
I rename the folder too, and I get this:

[3827:0609/165900:INFO:CONSOLE(134)] "Uncaught Error: Module did not self-register.", source: ATOM_SHELL_ASAR.js (134)

uropsm, 2015-06-17:
I have same issue. 

When I installed serialport, it created folder 'node-v14-win32-x64' instead of 'node-v43-win32-ia32

Windows7 64bit
node version : 0.12.4

kjlaw89, 2015-06-24:
Hey guys, apparently @usefulthink was able to get this to work by just changing how node-serialport built. I'm using it in my Electron setup right now and it's working well. Here's the link to the commit/fork:

https://github.com/usefulthink/node-serialport/commit/c5aace891834bd7b9b94f72e4420554721026a4f

a-giovani17, 2015-06-25:
I've tried @usefulthink's commit, it still show this error

Uncaught Exception:
Error: Invalid Access to Memory Location

The error comes from the binary (serialport.node), I wonder why.

But at least the folder problem is solved.

usefulthink, 2015-06-25:
I didn't really test anywhere except for OSX 10.10 which was all that I needed. But as all I did was replacing node-pre-gyp with node-gyp that is from my understanding unlikely to be the problem. 

Unfortunately, its several years since I last worked with Windows, so no idea..

usefulthink, 2015-06-25:
I couldn't find anything about it here, but that might cause such a problem - did you use `electron-rebuild` to recompile the add-on?

a-giovani17, 2015-06-25:
yes, I've used electron-rebuild but the error still appeared

voodootikigod, 2015-06-25:
Hey All, so the @usefulthink fork actually provided me with the insight into what is going with this and I would like to share it so everyone can understand it. 

Originally, node-serialport (this module) used a project called [node-gyp](https://github.com/TooTallNate/node-gyp) directly to provide the instructions for compiling the module on various platforms (Windows, Mac, Linux, etc.). While it worked (assuming all necessary dependencies) it was a nightmare as for doing any form of mass installation because undoubtably the necessary dependencies (build tools, compiler, etc) were never installed. This project, node-serialport, quickly became swamped with tickets about how to install on Windows, Mac, and Linux. 

Due to this, we moved to using a project called [node-pre-gyp](https://github.com/mapbox/node-pre-gyp), which allowed us to utilize pre-compiled binaries if they exist from a known location and fallback to the local build process if they do not. This was hooked into an elaborate build system using Travis-CI and AppVeyor to generate the necessary compiled versions for Windows, Mac, and Linux on each release published to npm.

Things were happy at this point since people using Windows, Mac, and Linux without the build components could easily use node-serialport without even being aware of its native module nature. 

During this time period, the project formerly named [node-webkit](https://groups.google.com/forum/m/#!msg/nwjs-general/V1FhvfaFIzQ/720xKVd0jNkJ), now [nw.js](http://nwjs.io), started to evolve and gain traction. It utilized its own derivative fork/hack from node-gyp in order to build the necessary native componets which was named nw-gyp and this choice had ramifications that [affected the compilation of other native modules](https://github.com/nwjs/nw.js/wiki/Build-native-modules-with-nw-gyp). Needless to say, as people began trying to use it in combination with the node-serialport project things did not work smoothly. Until the NW.js team published the previously linked instructions for node-gyp and node-pre-gyp, this project was blamed for incompatibility and many issues were filed. 

Then the [Atom](http://atom.io) editor was released. The Electron team that maintains the application framework around the Atom Editor updated to the [latest version of Chromium](https://github.com/atom/apm/issues/189) around September, 2014, which consequentially broke compatibility with any and all node-gyp compiled modules. Similar to NW.js, the Electron team [posted details](https://github.com/atom/electron/blob/master/docs/tutorial/using-native-node-modules.md) about how build Electron with other native modules, but those native modules mostly were not aware of this. As such, many issues were filed against this and other project concerned about the compatibility of that project when they try to integrate with Electron. 

The general node.js leadership has not openly discussed this mainly because they do not maintain end-user facing native modules, honestly there aren't a tremendous number of them. For those of us that do maintain a native module, the constant hamster wheel is enough to drive one insane. Until a maintainer can explore the other project, there is a very real sense that it could be the fault/issue/incompatibility of the project being maintained since Node.js, io.js and the ecosystem changes so quickly and its too much to keep up with. At this point, here is the general state of the world:

* **node-gyp** is the de-facto build standard for Node.js and native modules. All things should be compatible with node-gyp at a minimum or else it should be considered out of compliance.
* **node-pre-gyp** since it is an enhancement beyond node-gyp by adding pre-compilation and retrieval, it should be considered as part of the compliance suite for any native module.
* **nw-gyp** to my knowledge is only used by NW.js and as such should ensure that it or application that includes does not break compliance with node-gyp or node-pre-gyp. That is currently not the case however and inclusion of node-gyp and node-pre-gyp modules still require [special handling](https://github.com/nwjs/nw.js/wiki/Build-native-modules-with-nw-gyp).
* **Atom-Shell/Electron** is currently not compliant with node-pre-gyp due to the way that it identifies its node execution version and environment. That said, the node-pre-gyp maintainers are working on a way to make node-pre-gyp modules work with Electron. There is currently an [outstanding pull request](https://github.com/mapbox/node-pre-gyp/pull/112) that will provide support for this variation handling. Until that pull request is merged in, continue to use the [instructions](https://github.com/atom/electron/blob/master/docs/tutorial/using-native-node-modules.md) provided by the Electron team.

Specifically for node-serialport, as the prime maintainer I will not be merging in modifications that take this project backwards (ala reverting to node-gyp). Also, I will not be integrating special cases to support new native environments that have broken compatibility without thorough testing in every execution environment we already support. 

In summary, maintaining a native module in node.js is a constant battle. The node community needs to hold projects that break native compilation compatibility accountable by encouraging those projects to ensure node-gyp and node-pre-gyp modules can still be included without issue or failure.

usefulthink, 2015-06-25:
Thank you so much for writing this! Fully agree.

kjlaw89, 2015-06-25:
@voodootikigod I agree too, I think it makes sense for node-pre-gyp to resolve the issue on their side rather than you changing your project. In the meantime the work @usefulthink should get us by.

@usefulthink @a-giovani17 Windows compiling does work, as that's where I'm doing my build. I'm using the 2013 compiler "--msvs_version=2013", with electron-rebuild for compiling running the command ".\node_modules\.bin\electron-rebuild . --version=0.28.3". That downloads the 0.28.3 headers for electron and compiles against them.

Granted, I've done a lot of other things to my setup environment. If you wiped my computer and told me to get it all working again at this point I would likely be doomed. :P

voodootikigod, 2015-06-25:
@usefulthink just want to make sure the 'going backwards' is not a knock on you by any means (I like your solution in the short run :+1:). Cheers!

usefulthink, 2015-06-26:
@voodootikigod no worries, I just needed it to work and did the ugliest hack I could think of. I wouldn't have thought of making a PR of it anyway, and I agree with you that this would indeed be moving backwards for this project :)

0x80, 2015-07-04:
I am struggeling a bit to find out what the compatible Electron prebuilt version is and I suspect it is related to this issue. Are you using 0.27.2 because that is the last compatible version (uses iojs v1)? Or is serialport or the @usefulthink his fork also compatible with iojs v2 and Electron > 0.27? 

I have tested serialport with iojs-v2 (without electron) and that seemed to work just fine, but somehow in combination with electron I haven't managed to make it work with versions > 0.27

usefulthink, 2015-07-05:
Hi @0x80, you should be able to install my forked-version with `npm install serialport-electron`, then `electron-rebuild --force` should do the trick (i don't really know how electron-rebuild determines what to rebuild, but you will see the usual compiler-output when it does). This should work in every node/iojs-version that is supported by this module, as it will always rebuild from source. 
I guess you just have to try it.

0x80, 2015-07-05:
@usefulthink thanks! I've managed to make it work with 0.29.1. Your module didn't seem to be available in npm registry so I installed if from git.

0x80, 2015-07-05:
I'm still experiencing one major issue though. When I use serialport in combination with Electron, the CPU goes through the roof. The same code running on the same iojs version but as standalone only uses a very small percentage.  I already mentioned it here https://github.com/voodootikigod/node-serialport/issues/402. Any idea what might cause something like this? I am completely clueless.

ddm, 2015-07-16:
If anybody stumbles upon this while trying to use node-serialport with electron: there is a temporary solution in the form of the [serialport-electron](https://www.npmjs.com/package/serialport-electron) npm package.
Hopefully, in the long run, node-pre-gyp will support electron OOTB.

alexgoldstone, 2015-10-21:
I believe this should help but have yet to confirm: https://github.com/mapbox/node-pre-gyp/pull/175

vjpr, 2015-10-22:
@ddm I get the following message when I try to require `serialport-electron` from `main.js`:

```
Uncaught Exception:
Error: Module version mismatch. Expected 46, got 44.
    at Error (native)
    at Object.module.(anonymous function) (ATOM_SHELL_ASAR.js:137:20)
    at Object.module.(anonymous function) [as .node] (ATOM_SHELL_ASAR.js:137:20)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (/Users/Vaughan/dev-quantitec/intranav-manager/node_modules/serialport-electron/serialport.js:12:25)
    at Module._compile (module.js:434:26)
    at Object.Module._extensions..js (module.js:452:10)
```

What version of node and electron are you using?

ddm, 2015-10-22:
@vjpr I just tested it with iojs-v2.0.1

It works for the following versions:

    "serialport-electron": "1.7.4-electron-0.30.6",
    "electron-prebuilt": "0.30.6"

I believe it should work with 0.30.8 without a change. Just let me know if you want me to publish  `"serialport-electron": "1.7.4-electron-0.30.6"` to npm.

I think electron > 0.30.x requires a change to serialport to work properly because of a v8 version change.

vjpr, 2015-10-22:
Thanks!

You are right, changing version to 0.30.6 worked on Windows.

The regular `serialport` worked on OSX.

ddm, 2015-10-23:
@vjpr Glad it works :) Let me know if there is anything else I can help you with...

noopkat, 2015-10-28:
confirmed that I can use serialport v2.0.2 in electron prebuilt v34.0.1 on OSX Yosemite. Did need `electron-rebuild` to be installed and run additionally.

weiway, 2015-11-03:
@noopkat ```electron 0.34.2``` with ```serialport 2.0.2``` confirmed. Need to use ```node 4.x (v46)``` to install serialport module, otherwise the version number will not be correct```(5.0.0 v47)``` for Electron

noopkat, 2015-11-04:
@weiway good to know, thank you!

Mirodil, 2015-11-12:
same issue with 
`node 4.2.0`
`electron-prebuilt 0.34.2`
`serialport 2.0.2`

```
App threw an error when running [Error: A dynamic link library (DLL) initializat
ion routine failed.
E:\NOTE.JS\examples\dcAlram\app\node_modules\serialport\build\Release\node-v46-win32-x64\serialport.node]
```

Tibus, 2015-12-01:
I can use serialport v2.0.2 in electron v34 on OSX Yosemite, NodeJS 4.x but the CPU is very High (130%) in Electron. but with a simple code in standalone node that read every port at 115200 (like in my Electron app), the cpu is very low. So the probleme I think is when serialPort is in Electron?

dvlsg, 2015-12-07:
I ran into some new issues with serialport, electron v35, and windows. I managed to blindly stumble across a solution by using a combination of `electron-rebuild` to execute node-gyp and rebuild with electron v35 headers and a manual move of the serialport.node file from `node_modules/serialport/build/Release/electron-v0.35-win32-x64` into `node_modules/serialport/build/Release/node-v46-win32-x64`. It worked, but definitely seems wrong.

I definitely had to use the version from `electron-rebuild`, as well (probably since I have node v5 installed on my machine). I tried a lot of steps along the way, including grabbing the prebuilt version of v64-win32-x64 from the releases page as well as building serialport locally, but nothing worked other than the `electron-rebuild` file move detailed above.

weiway, 2015-12-14:
@Tibus Yes, that's a problem with serialport module. One workaround is to use the serialport in another node process

rafaelcorreiapoli, 2016-02-06:
Hi, I'm stuck with the problem "Expected 46, got 47"... but i'm using node v4.2.6 ... I installed serialport and ran electron-rebuild

palavrov, 2016-02-15:
@rafaelcorreiapoli this works for me. Tested on Mac (node.js 5.5.0) and Windows (node.js 5.6.0).

package.json
```JS
{
  "devDependencies": {
    "electron-prebuilt": "^0.36.7",
    "electron-rebuild": "^1.1.3"
  },
  "scripts": {
    "start": "electron app",
    "preinstall": "cd app && bower install && npm install && cd .. && electron-rebuild -v 0.36.7 -p -f -m app -w serialport -e node_modules/electron-prebuilt",
    "clean": "rm -rf node_modules && rm -rf app/node_modules && rm -rf app/bower_components"
  }
}
```
app\package.json
```JS
{
  "main": "main.js",
  "dependencies": {
    "async": "^1.5.2",
    "batch": "^0.5.3",
    "h5.modbus": "^0.1.0",
    "serialport": "^2.0.6"
  }
}
```

then to be sure that everything is clear
```bash
$ npm run clean
```
next step is instalation/compilation
```bash
$ npm install
```
finally the execution
```bash
$ npm start
```

Tibus, 2016-02-16:
@weiway but how to run another process of node inside a Electron app?

weiway, 2016-03-09:
@Tibus check this out: https://github.com/weiway/node-serialport-worker

Tibus, 2016-03-09:
Thank's @weiway, your node-seriaport-worker work really well and my CPU usage is normal now! 
Thank's verry much.

jacobq, 2016-03-17:
**TL;DR** Try running as administrator if `electron-rebuild` is failing.

@Mirodil, did you ever resolve the issue you encountered? I was fighting the same thing on my Windows machine. My project uses electron 0.36.11 (0.37.x seemed to have compile issues for me just like in [this issue](https://github.com/voodootikigod/node-serialport/issues/680) and environment is node v5.9.0, npm v3.7.3, and Windows 10 (64-bit) w/ Visual Studio express 2015 and Python 2.7.11 installed. I can successfully run `cd node_modules\serialport; node-pre-gyp install --build-from-source` and run my (Mocha) tests. However, when I would try to run electron it blows up with `App threw an error when running [Error: A dynamic link library (DLL) initialization routine failed.
\\?\C:\Users\Jacob\Documents\KFMT\kanolysis\node_modules\serialport\build\Release\node-v47-win32-x64\serialport.node]`.
[image]

I thought perhaps this is related to a node vs. electron incompatibility, so tried running `electron-rebuild --pre-gyp-fix -f` but it blew up with some errors. Upon closer inspection I noticed two things, one was that the name apparently changed such that I had to append `.cmd` and the other was that there was some kind of permission problem (why? I have no idea as I was doing all this in my home/Documents directory as my user). I then changed the git bash (MINGW64) shell shortcut to always run as an administrator....next time I ran that same `electron-rebuild.cmd` it succeeded and launched successfully in electron. Notably, there was a new `electron-v0.36-win32-x64` in the module's build/Releases/ folder in addition to the node one. (Hopefully this is helpful to someone else finding this....these problems can be overwhelming to someone not familiar with the Windows + node ecosystem)

jugglingcats, 2016-03-19:
I am about at the same point as @jacobq, and tearing my hair out!

Dependency versions:
    "serialport": "2.0.6"
    "electron-prebuilt": "0.36.7",
    "electron-rebuild": "^1.1.3",

Removed completely node_modules and ran npm install.

Serialport was successfully built using MS compiler to dir named node-v47-win32-ia32 (not sure why it is ia32 and not x64?).

Trying to run electron I have "DLL initialization routine failed", so it is finding the serialport.node file.

Trying to run rebuild_electron as admin I get the following output (node version v5.9.0):
```
npm run rebuild

> electron-cncp@0.1.0 rebuild C:\t
> electron-rebuild -v 0.36.7 -p -f -m app -w serialport -e node_modules/electron-prebuilt

spawn node ENOENT
Error: spawn node ENOENT
    at exports._errnoException (util.js:890:11)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:182:32)
    at onErrorNT (internal/child_process.js:348:16)
    at _combinedTickCallback (node.js:383:13)
    at process._tickCallback (node.js:407:11)

npm ERR! Windows_NT 10.0.10586
npm ERR! argv "C:\\opt\\nodejs\\node.exe" "C:\\Users\\alfie\\AppData\\Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js" "run" "rebuild"
npm ERR! node v5.9.0
npm ERR! npm  v3.8.0
npm ERR! code ELIFECYCLE
npm ERR! electron-cncp@0.1.0 rebuild: `electron-rebuild -v 0.36.7 -p -f -m app -w serialport -e node_modules/electron-prebuilt`
npm ERR! Exit status 4294967295
```

Any help much appreciated. Feels like I'm nearly there... just one hurdle to go ;)

jacobq, 2016-03-20:
@jugglingcats I think you're seeing the DLL error because the module that's there isn't the one you need for electron (i.e. the one `electron-rebuild` is supposed to make for you).
In case it's helpful, I was using node-serialport 2.0.7-beta1, electron-prebuilt 0.36.11, and electron-rebuild 1.1.3. I have 64-bit versions of NodeJS and Python 2.7.11 installed as well as Visual Studio 2015 Express w/ C/C++ support (though I'm not sure if any/all of these are needed). If I go into `node_modules\serialport\` and run `node-pre-gyp rebuild` or `node-pre-gyp install --build-from-source` it shows a few warnings but succeeds. Running `electron-rebuild` was the touchy one for me. I was getting `EPERM` (permission error). `ENOENT` just means "No such file or directory" so perhaps there is a problem with your path or a file/dir needed during the build process?

jugglingcats, 2016-03-20:
Eventually had similar results to @dvlsg. Had to rename the .node file but it is working. Thanks @jacobq for your input. Am not sure ultimately what was causing ENOENT but it went away at some point during my hackfest.

Installed Visual Studio Community 2015 with C++ support and the Windows 8 and 10 SDKs (was getting missing windows.h using the SDK tech preview mentioned elsewhere).

Uninstalled node win32 and installed node x64. Deleted global node cache and node_modules.

Fixed issue with node-gyp not finding python.exe by linking to python2.7.exe.

Ran electron-rebuild.

Renamed directory of node file from node_modules/serialport/build/Release/electron-v0.35-win32-x64 to node_modules/serialport/build/Release/node-v46-win32-x64.

Spun around 3 times and recited a silent prayer... and :tada:.

Windows 10
node: v5.9.0 (x64)
serialport: 2.0.6
electron-prebuilt: 0.36.7
electron-rebuild: 1.1.3
msvs_version = "2015"

I can only seem to require serialport from main.js, but am then passing it via remote.getGlobal() in my app script.

I am not seeing the high CPU issue (at least not in initial testing).

Hope this can help others, because this is one of my most painful experiences getting something running in quite a while ;)

ajfisher, 2016-03-30:
No dice for me I'm afraid - 

OSX Yosemite
node 4.3.1
serialport 2.0.6 and 2.0.2 tried
electron-prebuilt 0.36.7 and 0.34.1 tried
electron-rebuild 1.1.3

electron-rebuild appears to do something:

```
node-pre-gyp fixer: node_modules/gulp-watch/node_modules/chokidar/node_modules/fsevents/lib/binding/Release/electron-v0.34-darwin-x64 copied to node_modules/gulp-watch/node_modules/chokidar/node_modules/fsevents/lib/binding/Release/node-v47-darwin-x64
node-pre-gyp fixer: node_modules/serialport/build/Release/electron-v0.34-darwin-x64 copied to node_modules/serialport/build/Release/node-v47-darwin-x64
```

given the following command:

`electron-rebuild -v 0.34.2 -f -p -m node_modules/ -e node_modules/electron-prebuilt/`

Have tried electron-serialport too but not working either. Pretty much seeing the same error as everyone else where electron is expecting a different version than that done by the npm install.

Basically once app tries to connect it throws:

```
Error: Module version mismatch. Expected 47, got 46.
    at Error (native)
    at process.module.(anonymous function) [as dlopen] (ATOM_SHELL_ASAR.js:157:20)
    at Object.Module._extensions..node (module.js:450:18)
    at Object.module.(anonymous function) [as .node] (ATOM_SHELL_ASAR.js:157:20)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:313:12)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
    at Object.<anonymous> (/Users/ajfisher/dev/tmp/robotnik/node_modules/serialport/serialport.js:14:25)
    at Module._compile (module.js:425:26)
```

When the file is actually at:

```
/Users/ajfisher/dev/tmp/robotnik/node_modules/serialport/build/Release/node-v47-darwin-x64/serialport.node
```

Consistent behaviour on RPi as well (which is ultimately my target for this).

Any ideas much appreciated as I'm pulling out hair on this now.

weiway, 2016-03-31:
@ajfisher  Have you tried to use Node v5.x.x when rebuilding? And you might need to do some renaming.

reconbot, 2016-04-03:
https://github.com/OpenBCI/openbci-js-sdk/issues/31#issuecomment-205059018 is another case of electron-rebuild working out

ajfisher, 2016-04-03:
> OpenBCI/openbci-js-sdk#31 (comment)
> is another case of electron-rebuild working out

So node5 seems like a reasonable approach? I am on RPi so can control the
environment to a point so will give that a crack and see what happens.

reconbot, 2016-04-06:
Hi Everyone,
I moved the binaries around in beta4, I don't know if that's going to be a problem for any of you but I wanted someone who's using electron to give it a try. We also upgraded `nan` which I believe helps work with the newest electron.

Please try it out `serialport@2.0.7-beta4` and report back! #733

reconbot, 2016-04-08:
I've gotten a few reports that beta4 works great with `electron rebuild`. (and also for nwjs!) Can someone submit a docs patch for close #741? I'm not super familiar with electron but I think a lot of people would benefit from this threads collective experience.

Thanks!

techninja, 2016-04-08:
I'll see If I can't do it. Bug me here or in gitter if I fail to do so ;)

reconbot, 2016-04-08:
Also if you all could test `serialport@2.0.7-beta5` it's our release candidate. It's got a single change from beta4 around where the binaries go that should make your life a little easier when you rebuild.

techninja, 2016-04-09:
@reconbot Just tried out Beta5, Default node install works.. a bit messier in the build/Release folder, but it works. What still does not work, is that since there's only one binary output folder, `electron-rebuild` will refuse to step on it. You must delete `serialport.node` before rebuild will take over, but rebuild does work beautifully (and quietly!) after that.

Unfortunately for my use case I need to be able to have both of these binaries (Node.js terminal and Electron) available as I allow users to run serialport in just Electron, or in both Electron and a Node.js child process. This worked just fine in previous releases as I could actually keep [the binaries for all the OS' together](https://github.com/evil-mad/robopaint-build/tree/master/native_builds), and would then be able to also keep the v46 node v4.x built bins right next to the v47s for the latest node5 in Electron. Truth is if they use the same binary, I have yet to see the actual fault, as long as they use the same module version.

pimpaa, 2016-04-09:
@reconbot  `serialport@2.0.7-beta5` worked well for me with Electron 0.37.4, I just had to delete `serialport.node`.

Thank you.

reconbot, 2016-04-09:
That's annoying, good to hear though. I'll open an issue with them if there
isn't one already.

reconbot, 2016-04-10:
Are there other binary packages that work without needing this extra step of removing the binary?

ajfisher, 2016-04-10:
So I've just tried on my Mac plus an RPi, both running 5.10.x (using nvm) -
straight install using serial port beta 5 plus electron latest plus Johnny
five latest.

Worked fine on both platforms without an issue now it seems.

gohackfelipe, 2016-04-19:
Hi  guys!
Please... someone help me!

Using electron and serialport

When I run:

Error: Uma rotina de inicialização da biblioteca de vínculo dinâmico (DLL) falhou.
\\?\C:\projeto\meia-maratona-electron\node_modules\serialport\build\Release\serialport.node
    at Error (native)
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:158:20)
    at Object.Module._extensions..node (module.js:440:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:158:20)
    at Module.load (module.js:357:32)
    at Function.Module._load (module.js:314:12)
    at Module.require (module.js:367:17)
    at require (internal/module.js:16:19)
    at bindings (C:\projeto\****\node_modules\bindings\bindings.js:76:44)
    at Object.<anonymous> (C:\projeto\meia-maratona-electron\node_modules\serialport\serialport.js:16:44)

What can i do for this?

"electron-prebuilt": "^0.37.4"
"serialport": "2.0.7-beta5"
node version : 5.10.1
npm version : 3.8.6
python version: 2.7.11

reconbot, 2016-04-19:
@gohackfelipe Can you try `serialport@2.1.0` and then run `electron-rebuild` after installing?

gohackfelipe, 2016-04-19:
Hi @reconbot ..

I did what you asked for and did not work.. The same issue is happening...

Error: Uma rotina de inicialização da biblioteca de vínculo dinâmico (DLL) falhou.
\\?\C:\projeto\meia-maratona-electron\node_modules\serialport\build\Release\serialport.node
    at Error (native)
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:158:20)
    at Object.Module._extensions..node (module.js:440:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:158:20)
    at Module.load (module.js:357:32)
    at Function.Module._load (module.js:314:12)
    at Module.require (module.js:367:17)
    at require (internal/module.js:16:19)
    at bindings (C:\projeto\meia-maratona-electron\node_modules\bindings\bindings.js:76:44)
    at Object.<anonymous> (C:\projeto\meia-maratona-electron\node_modules\serialport\serialport.js:16:44)

reconbot, 2016-04-19:
Do you have the output of the rebuild?

gohackfelipe, 2016-04-19:
How get this output @reconbot ?

When i ran the `.\node_modules\.bin\electron-rebuild.cmd`, the power shell did not log anything..

[61 more replies]

closed 2017-01-19 by reconbot