serialport/node-serialport #540: Unable to change rts/dsr voltage

iwalkmyline, 2015-06-04:
Hello, I'm currently facing problems when I try to set rts/dsr to false. This is how I do it:

`s.set({rts:boo},function(e,b){});`

Where s is the SerialPort object and boo is a boolean.

My solution actually works with my laptop, but, when I move this to a MW810 computer (Motorola) it doesn't work anymore.

I've made one app in Java and another in VB and both work perfectly when I set rts/dtr to false but I really need to develop the solution using nodejs. Also, I use Realterm for testing and I can toggle rts/dsr signals.

Any suggestion or solution is appreciated. Thanks!

jacobrosenthal, 2015-07-07:
What OS and version and version of serialport for each?

iwalkmyline, 2015-07-07:
I found the solution: I was using an old version. Now I'm using the latest version. Thanks!

closed 2015-07-07 by iwalkmyline