serialport/node-serialport #541: multi data received

masnatacion, 2015-06-09:
I created my connection and send data succesfuly but I received the response in multiparts like this:

a0 13 01 89 94 30 00 39 36 30 36
31
37 36 31 35 32 00 00 6c 88

I will need  a0 13 01 89 94 30 00 39 36 30 36 31 37 36 31 35 32 00 00 6c 88

This is my connection

var serial = new SerialPort(me.path, {
      disconnectedCallback : function(){
      
      },
      baudrate: 115200
    }, false);

masnatacion, 2015-06-10:
Resolved

roryovercast, 2015-06-23:
I think I'm having a similar issue. What was the fix?

jacobrosenthal, 2015-07-07:
There is no guarantee to how data is emitted. Use a parser to only be notified when your delimiter is found (newline?) or use an array or something to buffer data i your handler

closed 2015-07-07 by jacobrosenthal