serialport/node-serialport #544: [linux] open file descriptor leak on serialport.open(), fd is not closed properly on serialport.close()

capricasechs, 2015-06-16:
On linux, when you open() and close() the serial port connection, the file descriptor to the device (e.g. to /dev/ttyUSB0) will not be closed properly. 
I've written a simple application that will open and close the connection to the serial port every second.
When you watch the file descriptors with 'lsof -c node' you will see that on every call to open() a new file descriptor is opened that will not be closed on close();

Used node version: 0.10.x
Used serialport module: 1.7.4

Example application to test the issue (it will open an close the serialport every second):

var serialport = require('serialport');
var portConnection = new serialport.SerialPort('/dev/ttyUSB0',
    {
        baudrate: 9600
    },
    false
);

var openClosePortEverySecond = function () {
    portConnection.open(function (error) {
        if (typeof error !== 'undefined') {
            console.log('failed to open serialport');
            console.log(error);
        } else {
            console.log('serialport opened');
            portConnection.close(function (error) {
                if (typeof error !== 'undefined') {
                    console.log('failed to close serialport');
                    console.log(error);
                } else {
                    console.log('serialport closed');
                    setTimeout(function() {
                        openClosePortEverySecond();
                    },1000);
                }
            });
        }
    });
};

openClosePortEverySecond() ;
process.stdin.resume();

You can watch the open file descriptoirs with: watch -n1 "lsof -c node | wc -l"

finim, 2015-06-16:
Update: it seems that this issue is only in the latest version(s), i've tested 3 versions:
1.7.4 : leaking 
1.7.1 : not leaking
1.6.2 : not leaking

gnomex, 2015-06-17:
I got the same issue...  I'm using node v0.12.4 and serialport: 1.7.4. The kernel version is `3.16.0-40-generic`.

I have tried your example code with `false` and next `true` values on `portConnection`, both had the issue.

koprda, 2015-09-04:
I got the issue ...
serialport: 1.7.4.

handle is open **2 times** in serialport_unix.cpp

1.
void EIO_Open(uv_work_t* req) {
  ...
  int fd = open(data->path, flags);

  if(-1 == setup(fd, data)){
  ...

2.
int setup(int fd, OpenBaton *data) {
  ...
  fd = open(data->path, flags)

reconbot, 2016-04-06:
We fixed a ton of bugs in our latest beta specifically ones addressing this issue. https://github.com/voodootikigod/node-serialport/issues/733 Please try it out `serialport@2.0.7-beta3` and report back!

reconbot, 2016-04-07:
I'm going to close this as we've confirmed this bug is fixed. Please reopen or make a new issue if you have more issues! Thanks for reporting it and tracking down the issue!

closed 2016-04-07 by reconbot