serialport/node-serialport #545: 250k baudrate on linux

Heash, 2015-06-25:
Now supported baudrates are: 115200, 57600, 38400, 19200, 9600, 4800, 2400, 1800, 1200, 600, 300, 200, 150, 134, 110, 75, or 50

Pyerial lib has patch for supporting 250000 baudrate. But it will be very nice if node-serialport will suport thit too.

jacobrosenthal, 2015-07-17:
PR Welcome. Wed need to investigate if that baud exists on osx and win as well.

ChrisTerBeke, 2015-07-21:
What is the status of this? We're really waiting for 250k baudrate support on linux. More and more devices use this baudrate to have faster communication nowadays (including a lot of IoT stuff).

jacobrosenthal, 2015-07-21:
PRs still welcome.

rwaldron, 2015-07-21:
@jacobrosenthal I was checking out the PySerial code and they just raise a ValueError for platforms that don't support. The patch for node-serialport should just throw a RangeError

ChrisTerBeke, 2015-07-22:
We've put some time in it to see what's going on and it seems very possible to do (latest linux kernels already support it), but sometimes the linux header files with correct baudrate info are missing so compiling throws an error on that. Also, the custom baudrate option in node-serialport seems interesting. Maybe we can adapt that functionality a bit to allow for more dynamic baudrates without having to compile for it (ofc we need to make sure that the clock dividers are set correctly)?

ChrisTerBeke, 2016-03-20:
We gave up on this and wrote our own drivers in C/C++ and made a node addon out of it.

farfromrefug, 2016-05-04:
@ChrisTerBeke you wrote an alternative to ```serialport```? is it open source?

ChrisTerBeke, 2016-05-04:
@farfromrefug it's kinda part of a larger application that is not open source, but if there's enough demand, we might look into open sourcing the serial comm part. It works on Linux, OSX and Windows (only confirmed on win10). Anyone interested?

reconbot, 2016-05-04:
We're going to land 250000 pretty soon. If you have any code that would help cleanup or improve node serialport, that would be wonderful.

reconbot, 2016-05-04:
Specifically, I've started refactoring the baudrate code, but have yet to land windows support for updating the baudrate. I'd love help on this PR https://github.com/voodootikigod/node-serialport/pull/763

farfromrefug, 2016-05-05:
@ChrisTerBeke If that s ok with you i think it would be awesome for you to share that project. Not to create a side project from ```serialport``` but to actually help improving it.
I am willing to help integrating this into a node module and into serialpart.
This is also an essential part of the product we are working on.

Thanks

superlou, 2016-05-07:
Very excited for 250K baud support for CNC applications!

ChrisTerBeke, 2016-05-07:
@reconbot awesome to heard 250k support coming soon! Any specific PR's you'd like some help with?
@superlou using it for similar purposes ;) check @printr3d

reconbot, 2016-05-08:
@ChrisTerBeke which platform are you looking to use?

We're using an older method of setting custom baudrates for linux/unix. And I have no idea how we're doing it for OSX, but it doesn't seem to work for 250000 on the ci, but it does on my mavericks machine. I don't know how to add windows support yet either, but it also doesn't work.

I've abstracted out how we set baudrates and put proper tests around it here https://github.com/voodootikigod/node-serialport/pull/763 for both opening and updating. This isn't something we can easily test on all of our CI's. We have "ports" available only on windows and osx (not linux) and the ports aren't actually physical devices so.. ¯\_(ツ)_/¯ 

Manual testing the branch on linux would help, docs and pointers to better ways to set the speed would help too. I've been cribbing from pyserial but the linux man pages are a bit vague and it doesn't cover windows afaict.

farfromrefug, 2016-05-09:
@ChrisTerBeke have you thought about the idea of sharing your module? I think that might help the serialport project a lot. ANd not only with the 250k support.
Thanks

reconbot, 2016-06-17:
This didn't make it into 4.0 but I promise you it's on my short list. Examples of 250000 baud on windows would be incredibly helpful.

reconbot, 2016-07-06:
Attached the patch for pyserial that added 250k support on linux

[pyserial.patch.txt](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/350019/pyserial.patch.txt)

fnipo, 2016-08-26:
Hi all,

I made a workaround to make it work in a debian linux.
I have changed the _serialport_unix.cpp_ using the _termios2_ structure, and I have made a one line change in the _uv.h_ node file of the _.node-gyp_ folder to make the module build works.
It is working for 250k baudrate communications with a Ultimaker 3D printer.
I did not test it extensively yet, but if you find it interesting I could send a Pull Request.

Kind regards,
Felipe Nipo.

reconbot, 2016-08-27:
Please do!

fnipo, 2016-08-29:
https://github.com/EmergingTechnologyAdvisors/node-serialport/pull/918

reconbot, 2016-11-05:
This blog post suggests copying the definition of termios2 into your own code as a workaround for not being able to have both structs available. http://www.downtowndougbrown.com/2013/11/linux-custom-serial-baud-rates/

I don't like that idea but it might work.

closed 2017-07-17 by reconbot

fluentart, 2017-07-17:
For some reason I can only get 9600 to work on mac/windows. I remember using node serialport a few years ago and was able to get 115200 to work back then, but not anymore. Anyone else with this issue? Perhaps I have to go to linux for nice 250k

reconbot, 2017-07-17:
Minding opening up a new issue for this? I'll need more background and details. With the upcoming `serialport@5.0.0-beta-8` I can confirm 250k works on windows, osx and linux. At least I think I can, you can help me ensure that.

commit 2595ef98 (2017-07-17): Test for 250k baud rate

closes #545
PR #1239 (2017-07-17): Baudrate.250000
Test for 250k baud rate

closes #545