serialport/node-serialport #546: Empty ports

matheusazzi, 2015-06-28:
I read all other closed issues related to no ports list, but this not solve my problem.

I'm on a Windows 8.1 with NodeJS v0.10.39
(I have tried with another NodeJS v0.10.x versions too)

Serialport v1.7.4

Have done all this:

```
Windows:
Install Visual Studio Express 2013 for Windows Desktop.
If you are hacking on an Arduino, be sure to install the drivers.
Install node.js 0.10.x matching the bitness (32 or 64) of your operating system.
Install Python 2.7.6 matching the bitness of your operating system. For any questions, please refer to their FAQ. Default settings are perfect.
Open the 'Visual Studio Command Prompt' and add Python to the path.
```

Still installed FTDI Drivers

I just create a simple app.js

```javascript
var serialPort = require('serialport');

serialPort.list(function (err, ports) {
  if (err) {
    console.log(err);
  }

  console.log('Ports: ' + ports.length);
});
```

And when I call `node app.js`

```
> Ports: 0
```

It's not finding any ports. Can someone help me?

matheusazzi, 2015-06-29:
On my os x, this are just listing 2 ports

```
Ports: 2
/dev/cu.Bluetooth-Incoming-Port
/dev/cu.Bluetooth-Modem
```

I'm having problem to list USB ports, please, there's something that am I missing?

jacobrosenthal, 2015-07-17:
If you're device cant be found on windows or osx, sounds like a device or computer problem sadly :( Did you find the issue?

matheusazzi, 2015-07-17:
Hi @jacobrosenthal, I didn't find the issue, probably I'm doing something wrong.

I'm just connecting 2 pcs through an usb male to male cable, I have tried with another cable too, so I think this isn't the problem.

jacobrosenthal, 2015-10-21:
Does your serial device show up as a com port in Device Manager?

jacobrosenthal, 2015-10-24:
Closed with no response from OP

closed 2015-10-24 by jacobrosenthal