serialport/node-serialport #547: OS X long delays between writes and multiple callbacks fired.

programmarchy, 2015-06-30:
I'm using serialport to connect to a bluetooth module on an AVR micro with the following code:

```
#include <avr/io.h>
#include <avr/wdt.h>
#include <stdlib.h>

volatile uint8_t bt_rx;

int main(void)
{
    // usart
    UBRR1L = 25; // 38.4k
    UCSR1A |= _BV(U2X1);
    UCSR1A |= _BV(TXC1);
    UCSR1B |= (_BV(RXEN1) | _BV(TXEN1));
    UCSR1C |= (_BV(UCSZ11) | _BV(UCSZ10)); // 8 bits, 1 stop bit

    while (1)
    {
        wdt_reset();
        loop_until_bit_is_clear(UCSR1A, RXC1);
        bt_rx = UDR1;
    }
}

```

All this does is consume bytes sent to it, pretty simple.

I also wrote a node program to dump a file through serialport:

```
if (process.argv.length !== 4) {
	console.log('Usage: node test COM FILE');
	return 0;
}

var SerialPort = require('serialport').SerialPort;
var fs = require('fs');

var comPort = process.argv[2];
var inputFilePath = process.argv[3];

var serialPort = new SerialPort(comPort);
var inputData = fs.readFileSync(inputFilePath);

serialPort.on('open', function (err) {
	if (err) {
		console.error('open', err);
	} else {
		console.log('open');
		serialPort.write(inputData, function (err) {
			if (err) {
				console.error('write', err);
			} else {
				console.log('done')
			}
		});
	}
});

process.on('SIGINT', function() {
  serialPort.close();
});
```

Then I dump a binary file with 20000 bytes: `node serialport COM4 input.bin`

On Windows, everything goes great. The output goes:

```
open
done
```

And I have a logic analyzer connected to the USART on my Bluetooth module:

[image]

Everything comes in nice and smoothly.

On Mac, it's a totally different story. The output goes:

```
open
done
done
done
done
done
done
done
done
done
done
done
done
done
done
done
done
done
done
done
```

And the logic analyzer has huge gaps between data:

[image]

Has anyone seen this before? Anyone know if this an issue with SerialPort, or OS X?

I am running OS X Mavericks.

programmarchy, 2015-07-02:
This looks like it could be a hardware issue. Tested on a Mid 2010 MBP with Yosemite, and everything was great. Late 2012 MBP has issues.

jacobrosenthal, 2015-07-17:
@programmarchy Any further updates? Was it that machine?

programmarchy, 2015-07-20:
Yes, it was the machine. Thanks for ping back.

Still haven't figured out the problem with the machine, but the same exact code worked on another Mac of the same model.

Closing.

closed 2015-07-20 by programmarchy