serialport/node-serialport #548: Will there be an api to track device addition or removal?

opensourcegeek, 2015-07-01:
Currently I'm looking at ways to make any device addition or removal event to be bubbled up to nodejs. I'm interested only in devices which come up as serial devices, and I was wondering if that sort of API will be supported? Mainly it would be good to abstract OS specific code that does the tracking of devices - RegisterDeviceNotification like function in Windows, udev on linux for example. Any thoughts?

jacobrosenthal, 2015-07-17:
Sadly we're kind of in feature lock down around here with our original maintainer out of commission for a while, and while we wait for some lower level node changes to better support serialport builds across platforms. That said if you come up with something let us know.

closed 2015-07-17 by jacobrosenthal

opensourcegeek, 2015-07-17:
I have written a separate library using nan/libuv to surface code written by my colleague to do device notification in windows which notifies addition/removal of composite devices to nodejs. It's a callback based API. Though it's made for composite devices, I can probably make looking up drives associated to a serialport optional. I can use udev to do the same thing in linux too and I'd have to do this anyway. But I have no experience building for macos. I think this code can be merged into serialport library to setup tracking of devices rather than publishing it as a separate library. 

The API currently takes a vid/pid, a callback for device addition and a callback for device removal. The callbacks are called with a device object which has various low level details along with serial port the device is enumerated as and mounted drive name for device attached. 

Would this feature be of any interest? If it makes sense to include it into node-serialport then i can spend time to make it work with it.

jacobrosenthal, 2015-07-17:
Were actually kind of in feature lockdown as were currently in a node working group looking at using libuv or something else to remove most of our compiled code. I could see the emitters/whatever living in there and us utilizing those new apis. Can you get with them and see what they think?